letscontrolit/ESPEasy #598: Problem with MQTT and Home Assistant

leobel96, 2017-11-21:
Hi to all, I'm having some problems with MQTT and Home Assistant. I've installed the last version of ESPEasy (https://github.com/letscontrolit/ESPEas ... .0.0-dev12) on my NodeMCU v1.0, I've given a name to it (https://imgur.com/SGikeGy) and I've configured MQTT to work with Home Assistant (https://imgur.com/KoISxCd). Then I've tried to make a simple MQTT request (https://imgur.com/C2ju6Ti) from HA to EE but nothing happens: the pin remains HIGH. I've also tried adding and removing "/" at the beginning or changing the Unit name or sending a "1" instead of a "0" but nothing changes. Home Assistant is correctly configured because I'm using the Embedded Broker and I've already tested it with other MQTT clients.

Grovkillen, 2017-11-21:
Take this to the forum.

closed 2017-11-21 by Grovkillen

leobel96, 2017-11-21:
On the forum I received no answer...

TD-er, 2017-11-21:
Can you give a link to the topic?

leobel96, 2017-11-21:
Now @Grovkillen has answered me. Anyway this is the link: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3789&p=20929#p20929