letscontrolit/ESPEasy #599: 1.2.4rc2 chunked transfer fails

s0170071, 2017-11-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
compile the ESPEasy using the 2.4.0rc2 Arduino library, open the webinterface. You will find the size of the chunks printed in plain text. See also here: 
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3673

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes.

### Expected behavior
The web interface should of course not have these numbers.

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:
Wemos D1mini

Software or git version:
Ardunino core 2.4.0rc2 and ESPEasy 2.0.0 r12

Solution: 
The 2.4.0 does chunked transfer implicitly, so no need to do it manually.

psy0rz, 2017-11-23:
Oh how annoying..all this effort to work around those bugs in the first place:)

closed 2017-12-03 by mvdbro

s0170071, 2017-12-03:
This fixes it, but I am afraid it is temporary until 2.4.0 is released. It would be better to do chunking only if compiled with ESP8266 and 2.3.0  and dont do it in any other case- no matter if ESP32 or ESP8266.

commit 83abac8c (2017-12-03): Fix #599