letscontrolit/ESPEasy #600: Using 2.4.0rc2 web interface hangs and crashes with VPN

s0170071, 2017-11-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Use Arduino core 2.4.0rc2 and access the ESPEasy web interface with an Android device via VPN

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes.

### Expected behavior
Should work.

### Actual behavior
Websites are often not working (only small ones) and the ESP crashes. Probably sth with the mtu size. Packets seem to clog up the stack subsequently.

### System configuration

Hardware:
Wemos D1 mini

Software or git version:
ESPEasy 2.0.0 r12 compiled with Arduino core 2.4.0rc2

### workaround:
a) insert a delay of 20ms after each WebServer.sendContent() to wait until the previous packet is sent
or
b) monitor the free heap memory and wait for it to return after  WebServer.sendContent() in combination with a timeout. Working code example below (uses implicit 2.4.0rc2 chunking).

`void sendWebPageChunkedData(String& log, String& data)
{
	String tmp;
	delay(20);
	checkRAM(F("sendWebPageChunkedData"));
  if (data.length() > 0)
  {
    statusLED(true);
    log += F(" [");
    log += data.length();
    log += F("]");
    String size;
    size=String(data.length(), HEX)+"\r\n";
 
    //do chunked transfer encoding ourselfs (WebServer doesnt support it)
	#define MAXSIZECH 1300
	int i = 0;
	int memory = 0; 
	int timeout = 0;

	// split into chunks of MAXSIZECH
	for (int l = 0; l < (data.length() / MAXSIZECH);l++) {
		tmp = data.substring(i, i+ MAXSIZECH);			// prepare tmp string for passing to sendContent()
 		
 		memory = ESP.getFreeHeap();					// available memory before send
		timeout = 0; 								// init timeout
		WebServer.sendContent(tmp);
 		i = i + MAXSIZECH;
		while ((ESP.getFreeHeap() < memory)&&(timeout <50)) {	// wait for memory to return or the timeout
			delay(5);	
			timeout++;				
			}
		}
		
	// send remainder	
	tmp = data.substring(i, data.length());
	timeout = 0;
	memory = ESP.getFreeHeap();					// available memory befor send
	WebServer.sendContent(tmp);
	while ((ESP.getFreeHeap() < memory) && (timeout <50)) {	// wait for memory to return
		delay(5);
		timeout++;
	}

    data = F("");   //free RAM
    tmp = F("");
  }
}

void sendWebPageChunkedEnd(String& log)
{
  log += F(" [0]");
   WebServer.sendContent("");
   WebServer.client().flush();
   WebServer.client().stop();
}
`

TD-er, 2017-11-23:
I think checking the heap during transfers is always a good idea, not only because of these VPN issues.
Tunneling may add some bytes to the package size. Therefore it may also be a good idea to check if lowering the MTU to 1492 bytes is working.

Also it may be a good idea to look at the way Strings are being used.
This may cause a lot of heap fragmentation, but that's probably big enough for a separate issue :)

psy0rz, 2017-11-23:
Hmm thats strange...the esp shouldnt be affected during sending right? Or is the mtu adjustable in espeasy?

This should like a more serious bug perhaps. Might need to move it to 2.0 in that case.

s0170071, 2017-11-23:
Whats more, the sendWebPageChunkedData function needs to ends strings of up to 8kB. 
I was thinking that is may be a good idea to start sending the webpage before the string gets that large.
Since for chunked transfer no prior knowledge about the size of the website is required, this could work.

s0170071, 2017-11-25:
I created an issue in the core git but I think this not getting anywhere.
https://github.com/esp8266/Arduino/issues/3871

s0170071, 2017-11-25:
Latest git of the core fixed the issue. Presumably due to switching to lwip2.

closed 2017-11-25 by s0170071

s0170071, 2017-11-26:
Quick follow up. When compiling with the latest core version do the following: 
 install Arduino IDE in portable version (after unpacking .zip file create folder named "portable" and run Arduino.exe - if you need more information STFW ( https://www.arduino.cc/en/Guide/PortableIDE )) and try to install Github version into this one. Do not install any other Core via Board manager. (Thanks to Pablo2048)
In the Arduino IDE choose Tools/LwIP variant->Espressif (Xcc). This fixes the issue.
I also tried v2 but this one fails due to a "lwip/tcp_impl.h" header which cannot be found.