letscontrolit/ESPEasy #601: %ip1% %ip2% %ip3% %ip4% missing

s0170071, 2017-11-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
this is more a feature request ;-) but no big deal.
%ip% prints the whole IP whis then drops the most interesting fourth number on a small oled display. 

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)

### Expected behavior
%ip1% should return the first number of the ip address. %ip2%...

### Actual behavior
not implemented

### System configuration

Hardware:
Wemos D1 mini
Oled display 64x48 pixels

Software or git version:
Arduino core 2.4.0rc2
ESPEasy 2.0.0 r12

Grovkillen, 2017-11-23:
I see the use for this. I added it to "future" but some dev. might implement it faster.

s0170071, 2017-11-23:
I have done it already (was really easy) but not committed it yet. 
It will be my first github commit, so I need to figure things out first...

psy0rz, 2017-11-23:
thanks, i've merged it :)

closed 2017-11-23 by psy0rz