letscontrolit/ESPEasy #603: LCD output not correctly displayed 4 x 20 display

supertrerkker, 2017-11-23:
If I fill in the lines available within ESP mega Easy for the device Display - LCD2004 with any information on line 4 it's displayed on line 2. ;-). Also if the info on line 2 is more then 20 characters the overflow is displayed on line 4. Line 1 and 3 are correctly displayed.  I'm using a 4 x 20 lines display.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes).  
Yes it is persistant. No changes after powercycle or reset.

### System configuration
The display is connected to a esp12e with the latest ESPeasy v2.0.0-dev12. Domoticz is running on the latest stable version 3.8153.

TD-er, 2017-11-23:
Could you please check it also with the latest code in the branch?
A recent fix did also set the device pointer to NULL.

I see at least one bug in the code.
When you change the size parameter, the display device pointer should be destroyed and re-initialized.

supertrerkker, 2017-11-23:
Gijs,

I'll give it a try tomorrow.

Regards,

Henk.

soif, 2018-02-13:
@supertrerkker would you please confirm that it is fixed in this commit, so that one could close the issue 👍

supertrerkker, 2018-02-15:
Hi,

Looks fine now thanks.

;-)

closed 2018-02-15 by supertrerkker

soif, 2018-02-15:
You're welcome :-)

PR #852 (2018-02-13): P012 - Fix/handle lines exceeding LCD width + allow to clear line b4 print
1) Update the Liquid Crystal I2C lib to the latest one

2) Fixes #603 

3) Add a **Mode** parameter which handle how the message received from LCD command are handled :

-  Text exceeding the LCD size will nicely continue on the next line, defaut behavious in v1.4, but it was bugged  (see #603)
-  Text exceeding the LCD size are truncated at the end of the line (without continuing on the next lines)
-  Same as 2, but the portion where the message is written, is first filled with space ( a features asked on the forum, an [... cut]