letscontrolit/ESPEasy #604: 2.4.0rc2 percent decoding crashes ESP

s0170071, 2017-11-23:
### Steps to reproduce
compile with 2.4.0rc2 and try to use a %systemvariable% in the web form, e.g. the display

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes
### Expected behavior
It should work like in 2.3.0

### Actual behavior
the string that I enter in the form gets truncated after the %. Whats more, it is likely to crash the ESP.
The behaviour is related to the url percent encoding. 
If I want a line 
`bla.exe?x=1+dom.GetObject("%valname%").State(%value%) `
in the form, I need to enter 
`bla.exe%3Fx%3D1%2Bdom.GetObject%28%22%25valname%25%22%29.State%28%25value%25%29`
for it to work. 

Solution: 
I have done some digging. It seems that 2.4.0rc2 has fixed a bug (or at least they made a change) that automaticly percent decodes strings received from html <form  method='post' > forms. If the form is a <form  method='post' enctype='multipart/form-data'> it all works fine. The string is then not encoded by the browser and not decoded by the ESP. I tried it and it works for me (firefox on win7 and chrome on android). As I am not an expert for web coding it would be nice to get some confirmation.

What remains a mystery is why the ESP crashes with %systemvariables% the <form  method='post' > forms. If the ESPEasy does not work with malformed form strings this should be fixed.

### System configuration

Hardware:
wemos d1 mini

Software or git version:
2.4.0rc2 core and 2.0.0r12

NigelTwo, 2017-12-17:
Device formulas might be affected by this too. 

I could not enter a formula for the "Analog Input" Device.
Environment:  FireFox 57.0.1 (Linux) and  ESPEasy 2.0.0(dev12) made under Arduino IDE (1.8.3) with ESP8266 Core Version:  2_4_0_RC2.

What I tried to enter in the Device form was: %value%/100, but it just disappeared on submit.
What I entered (after reading this post) is: %25value%25%2F100, and now it is persistent (and computes).

Thank you for discovering this.

s0170071, 2017-12-17:
I am not even sure that this is only core related. Just in case it is- Someone should open a bug report on the core here: https://github.com/esp8266/Arduino/issues?
This would would require
1. the justification why it has to work with `form method='post'` and not with `enctype='multipart/form-data`
2. a sample code and you can hardly drop them the whole ESPEasy
3. leave a note here.

I am stuck at point 1 and fixed the issue for myself by replacing all `form method='post'` and not with `enctype='multipart/form-data`. Do it with a #define - that gives you some flexibility to play with.

Grovkillen, 2017-12-18:
Ok, i changed it to "bug" since we still  don't know if its "our" problem or not.

s0170071, 2017-12-21:
filed an issue here: 
https://github.com/esp8266/Arduino/issues/4013

s0170071, 2017-12-30:
It is core related indeed.
https://github.com/esp8266/Arduino/pull/3313 fixes the issue.

closed 2017-12-30 by s0170071