letscontrolit/ESPEasy #609: Switch Inversed Logic error

melwinek, 2017-11-28:
I have a relay LOW LEVEL TRIGGER.
I turn on Inverse Logic in SWITCH options.
After executing: http://192.168.3.11/control?cmd=GPIO,0,0 my relay is On and Value in devices page is 1.
After executing: http://192.168.3.11/control?cmd=GPIO,0,1 my relay is Off and Value in devices page is 0.

It looks like the web works fine with Inverted Logic, but the web commands are not working properly.

Grovkillen, 2017-11-28:
Well, the "inverted logic" is applied on-top of the gpio logic. You suggest that the real gpio logic should be override by the "inverted logic"? Sounds like a bad idea.

I suggest you use rules to do what you want.

Grovkillen, 2017-11-29:
If no one else is giving me an reason to not close this by tomorrow I will close it.

Thanks for the update though.

sassod, 2017-11-30:
@Grovkillen I think you should close this as this is the way it should be,
inverse logic is there for inverse ON/OFF, web page should state in what
state is the relay (or any other device) and not GPIO.
If the web status is same as GPIO than it is just logic ON/OFF without
being inverted.
my 50 cents on the matter

closed 2017-11-30 by Grovkillen