letscontrolit/ESPEasy #610: Generic HTTP Advanced Variable is not casted

kickbutts, 2017-11-28:
### Steps to reproduce
Send a http post to a server and use a variable in the body. The Variable is not casted into a number.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes
### Expected behavior
Send a String to the Server in which a variable is casted into a number (to be precise into a double).

### Actual behavior
The System just sends the Body like this:
`Flur value=%DHT22#DHT22Temp%`
For working correct, the variable %DHT22#DHT22Temp% should have been cased into the actual sensor reading.
The server (influxdb here) rejects the value, because it expects a double but get a string.

### System configuration

Hardware:
WeMos D1 Mini

Software or git version:
v2.0.0-dev12

Grovkillen, 2017-11-28:
[DHT22#DHT22Temp] is the correct syntax to use.

kickbutts, 2017-11-29:
Same Problem.
`{"error":"unable to parse 'Flur value=[DHT22#DHT22Temp]': invalid boolean"`
I tried this before.

TD-er, 2017-12-06:
I am not sure how to reproduce this bug.
Is it in the "Generic HTTP" controller?

As test I let it send the WiFi strength:
`Dec  6 23:26:05 ESP Unit: 31 : /demo.php?name=Breadboard%5fESP&task=WiFi&valuename=WiFi&value=-65.00
`
That looks just fine to me.
Or is it somewhere else?

kickbutts, 2017-12-13:
Yes it is the "Generic HTTP Controller".

kickbutts, 2017-12-13:
Just read #635 . How do i install the new version? Do i have to wait for a new release of the developer version?

TD-er, 2017-12-13:
It has not yet been merged to the LetsControlit branch.
I think @psy0rz will start merging any day now (more likely in the weekend) and hopefully the 2.0.0-dev13 will appear soon.
Maybe tomorrow I will create a merge in my fork just to get a testversion myself to test all the commits I have on my own devices and then I can share some binaries, just for testing purposes.

kickbutts, 2017-12-13:
Okay. Got it thanks.

psy0rz, 2017-12-18:
we're currently merging a fix that might be related, it will be in v2.0.0-dev13, please check if the problem still persists in this version.

Grovkillen, 2017-12-19:
We need to test this asap and close the issue if it is resolved.

kickbutts, 2018-01-03:
Tested the new release and it did not change anything.
I captured the network traffic at my influxdb server and this is what arrived at the server:
```
POST /write?db=keller_temp HTTP/1.1
Host: 192.168.178.46:8086
Connection: close
Content-Type: application/x-www-form-urlencoded
Content-Length: 37
Vorratskeller value=[DHT#Temperature]
```
So still the same problem. When I change to a fixed value like 24 everythings works fine.

TD-er, 2018-01-03:
I will have a look at this later this evening.
Could you perhaps share some screenshots of the configuration of your Generic HTTP Controller?

kickbutts, 2018-01-03:
[image]
The Configuration is simple.

TD-er, 2018-01-03:
OK, there are apparently 2 Generic HTTP plugins, C008 and C011. You're using C011.
Please set the log level to 4 (or higher) in Tools => Advanced => Log Settings.
Then there will be log entries:

- "HTTP before parsing: "
- "HTTP after parsing: "

And it looks like the values have to be defined like this:
%1%%vname1%____%tskname%____%val1%%/1%

%1%...%/1% are delimiters here.

Taken from the sourcecode:
```C++
// Example:
// %1%%vname1%____%tskname%____%val1%%/1%%2%%__%%vname2%____%tskname%____%val2%%/2%
// will become in case of a sensor with 1 value:
// SENSORVALUENAME1____TASKNAME1____VALUE1  <- everything not between %1% and %/1% will be discarded
// in case of a sensor with 2 values:
// SENSORVALUENAME1____TASKNAME1____VALUE1__SENSORVALUENAME2____TASKNAME2____VALUE2

// example string:
// write?db=testdb&type=%1%%vname1%%/1%%2%;%vname2%%/2%%3%;%vname3%%/3%%4%;%vname4%%/4%&value=%1%%val1%%/1%%2%;%val2%%/2%%3%;%val3%%/3%%4%;%val4%%/4%
%1%%vname1%,Standort=%tskname% Wert=%val1%%/1%%2%%LF%%vname2%,Standort=%tskname% Wert=%val2%%/2%%3%%LF%%vname3%,Standort=%tskname% Wert=%val3%%/3%%4%%LF%%vname4%,Standort=%tskname% Wert=%val4%%/4%
```

It is not really clear from the sourcecode what the syntax should be and I think this really needs some clarification in the documentation.
So please test using the debug level on your debug output (serial, log server, web log) to see what will happen.
For inspiration on what strings to use, see the sourcecode starting at this line: https://github.com/letscontrolit/ESPEasy/blob/e4d68b9d0e7a477cd57ae3d6eabc2265b87b5964/src/_C011.ino#L310

psy0rz, 2018-01-04:
There is documentation here: https://www.letscontrolit.com/wiki/index.php/EasyProtocols#Generic_HTTP_Advanced

kickbutts, 2018-01-04:
Ah okay. Got it. So there is no bug in the software, the problem is the did not understand the documentation.
Thanks for your help!
I will do some testing and maybe i am able to write a clearer documentation. If this is the case to whom can i send it?

closed 2018-01-04 by kickbutts

Grovkillen, 2018-01-04:
Please send anything you want to be uploaded to wiki to either me or
@ShardanX