letscontrolit/ESPEasy #613: FR: PCA9685 plugin should be able to change pwm frequency

M0ebiu5, 2017-12-01:
There should be a possibility to change the pwm frequency of the pca9685 chip.

TD-er, 2017-12-02:
I ordered such a board to make the change and be able to test it. 
Only thing is... now I have also an excuse to order something like a scope to display the waveform ;)

beicnet, 2017-12-02:
@TD-er I had those excuses a few times too!!! :D

M0ebiu5, 2017-12-02:
I'm glad that i could help you with the scope :)
Thanks for implementing the new feature!

Here is a snippet from peter scargill - dont know if you need it or if its any help:
`void IFA pwmFrequency(uint8_t chipAddr, float freq)
{
PWMAddr=chipAddr;
pwmWrite(PCA9685_MODE1,0); // saves a reset function

freq *= 0.9;  // Correct for overshoot in the frequency setting
float prescaleval = 25000000;
prescaleval /= 4096;
prescaleval /= freq;
prescaleval -= 1;
uint8_t prescale = ifloor(prescaleval + 0.5);
uint8_t oldmode = pwmRead(PCA9685_MODE1);
uint8_t newmode = (oldmode&0x7F) | 0x10; // sleep
pwmWrite(PCA9685_MODE1, newmode); // go to sleep
pwmWrite(PCA9685_PRESCALE, prescale); // set the prescaler
pwmWrite(PCA9685_MODE1, oldmode);
os_delay_us(5000);
pwmWrite(PCA9685_MODE1, oldmode | 0xa1);  //  This sets the MODE1 register to turn on auto increment.
}`

M0ebiu5, 2017-12-18:
Solved with PR #638

closed 2017-12-18 by M0ebiu5