letscontrolit/ESPEasy #617: Azure controller support

Grovkillen, 2017-12-03:
https://github.com/Azure/azure-iot-arduino-protocol-mqtt

I have some friends who is in loop when it come to more industrial side of sensoring / IoT. I got the feeling that Azure support out of the box is something that would expand the user base a lot. The Azure world has a lot of tools for big data and ESP Easy would fit into that world when it comes to input data.

TD-er, 2017-12-03:
Do you have any documents on that?
And can you run an Azure environment on a local PC, without any subscription?

Azure is some kind of group name of several products. One of our clients at my job, wanted to move the product we developed for them, to Azure.
In the end it was just moving to an Ubuntu instance running in a virtual machine inside Azure. Just to show that "Azure compatibility" can be interpreted in so many ways.

Grovkillen, 2017-12-03:
You need to support "MQTTS" and communicate over 8883. The connection string is quite long and that need to be supported in the publish library as well. The reported format should be JSON or other easy-to-parse format.

https://docs.microsoft.com/azure/iot-hub/iot-hub-sparkfun-esp8266-thing-dev-get-started

It's the IoT Hub that I talk about, not using Azure as cloud deployed "old fashion server".

https://azure.microsoft.com/services/iot-hub/

TD-er, 2017-12-15:
Still leaves my other question: 

> And can you run an Azure environment on a local PC, without any subscription?

Otherwise it is very hard to test/debug.

Grovkillen, 2017-12-15:
True, only online as far as I know.

giddyhup, 2017-12-30:
I would reword the request to ask for generic MQTTS support. That way other offerings (e.g. AWS's) may be covered as well.

bernharddisselhoff, 2018-01-01:
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support gives you all the information needed.
It should be enough to move to arduinomqtt which supports mqtts TLS/SSL.
Azure IoT HUB is only online availible with subscribtion. There is a 8000 mesg a day version of IoT HUb for free availible.
ESPEasy would really have an advantage  to support AZURE IoT Hub because so many people are using ESP8266.
If you need someone for testing let me please know.

psy0rz, 2018-01-02:
People watching this: Vote up the features or bugfixes you want like to see fixed first. Use the thumbs UP arrow for that on the first comment. This way we can determine where to start, because we have LOTS of requests. :)

mstjerna, 2018-02-11:
I would love a more flexibel mqtt/tls support with longer Id/USER/pass

TD-er, 2018-02-11:
TLS is somewhat of a problem.
Not really for CPU power, but support for it in the used Arduino ESP8266 core.
I tried some for the email notification part, but the ESP kept crashing.
In Arduino core 2.4.0 library it is supposed to be working, but that core has other issues with regard to the webserver part. (chunked encoding errors)
So we have to fix those parts first before we can do SSL/TLS.

Do you have any idea on how long id/User/Pass should have to become?
Increase of those is also an issue, due to the way those information chunks are stored in SPIFF. But others have also requested longer strings for those.
So I really like to have an idea on how long they should be. See #822

bernharddisselhoff, 2018-02-21:
A typical azure connect string looks like:
HostName=bedesp2017iot1f5b5.azure-devices.net;DeviceId=geraet100;SharedAccessKey=V5vnXLOTjqSDyWM12mV0PSIW4L/MreaTkKIbG5rAfLE=

String containing Hostname, Device Id & Device Key in the format:
"HostName=<host_name>;DeviceId=<device_id>;SharedAccessKey=<device_key>"