letscontrolit/ESPEasy #620: ESPeasy "DEAD" after upgrading deom dev 11 to dev 12 :(

thundergreen, 2017-12-03:
Hey guys.

just upgraded via OTA the new dev 12 firmware and my esp won't show LCD dosplay nor connect somehow to the network.

What changed that it crashed like this ?

TD-er, 2017-12-03:
Did you also push the reset after flash?

I have quite often that the ESP won't boot the first time after flashing.
However it isn't an OTA flash I am doing here, but just via the USB cable.

A single press on the reset button (NodeMCU boards here) is enough to get the board started again.

Grovkillen, 2017-12-04:
Yes, try @TD-er suggestion and take stuff like this to the forum.

closed 2017-12-04 by Grovkillen

thundergreen, 2017-12-04:
OK folks! will do a fresh flash .. reset button and will see if it works .. thanks so far

thundergreen, 2017-12-04:
But one silly question is left:

after flashing.. are there still my previous settings stored somewhere? I assume only flashing the Firmware not the memory itself?

TD-er, 2017-12-04:
When you use the same type of firmware (memory size for example), the settings will be kept

thundergreen, 2017-12-04:
Well I just tried ota ..did not work.. Then via USB and I saw the access
point ..could connect but web interface was not reachable... I had dev 11
before..

TD-er, 2017-12-04:
Could you open a topic about this on the forum? At Github pages you don't get as many views as on the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1).