letscontrolit/ESPEasy #623: InfluxDB / Telegraf "controller" needed

Grovkillen, 2017-12-04:
If we want to report to a InfluxDB without having to have a controller in-between we need to output the data according to this JSON format:

https://www.influxdata.com/integration/mqtt-monitoring/
https://github.com/influxdata/telegraf/blob/master/docs/DATA_FORMATS_INPUT.md

*Telegraf is a Agent to read (in this example) MQTT and feed that into the database.

https://www.influxdata.com/time-series-platform/telegraf/

LeeNX, 2018-07-20:
@Grovkillen - This is not just output MQTT messages as JSON strings?  
  
I had a similar need and ended up having to create another mqtt section for my ESPEasy devices and use ```data_format = "value" and ```data_type = "float"``` as follows
```
 [[inputs.mqtt_consumer]]
   servers = ["tcp://127.0.0.1:1883"]

#   ## MQTT QoS, must be 0, 1, or 2
   qos = 1

 ## Topics to subscribe to
  topics = [
    "/data/ESP_Easy_Gate/#",
    "/data/ESP_Easy_BMP280/#",
  ]

#    data_format = "json"
  data_format = "value"
  data_type = "float"
``` 
This mostly works.  
  
I was going to request a feature to output mqtt as json values, which I can give an example of?

LeeNX, 2018-11-08:
Now that #1684 is mostly sorted out, now looking at the next part of my setup. How does MQTT in JSon coming?

Grovkillen, 2018-11-08:
We want to release a stable and let us concentrate on next steps without users finding a nightly release corrupt their system. We need to have to redo some major parts of the firmware.

LeeNX, 2018-11-08:
@Grovkillen - Thanks.  
  
I am for stable release too!  
  
Can we add a label or something? Noted for work on after release.

clowrey, 2019-01-25:
I also would really appreciate direct to InfluxDB controller - just one less thing that has to be installed and running on server to monitor data

haldi4803, 2019-10-21:
Same here...
Would be pretty cool to directly write into database instead of having one more thing to install.