letscontrolit/ESPEasy #630: iRAM usage for plugins using SoftwareSerial

TD-er, 2017-12-08:
I noticed that all plugins using SoftwareSerial, use 380 Bytes of iRAM.
And since we are almost running out of iRAM (current 32364 of 32768 max), it may make sense to try to keep this down a bit.
Current plugins using this:

- P049_MHZ19
- P052_SenseAir
- P053_PMSx003
- P065_DRF0299_MP3
- P071_Kamstrup401

My first question is, is it correct to assume the Software Serial module does indeed add about 380 Bytes of iRAM usage?

Do all these plugins need their own SoftwareSerial object?

How many of these can be active concurrently?

Does it make sense to add some serial layer class, which can be re-used?

papperone, 2017-12-09:
+1 and probably other plugins use it from the playground (like my _P171_PZEM-004T)

TD-er, 2017-12-09:
I created a pull request for this.
Hopefully the iRAM usage is now lower.
This should be tested thoroughly with >1 software serial plugin.

TD-er, 2017-12-10:
Like I mentioned also with the pull request (which is currently still a work in progress), I would rather label this as a 2.1 feature, not 2.0

psy0rz, 2017-12-18:
yep needs to go in 2.1.0.

i think some plugins already create the SoftwareSerial object dynamicly, exactly for this reason.

It does indeed use valuable iRAM or IRAM, since its not an apple product :P

closed 2018-03-04 by TD-er