letscontrolit/ESPEasy #634: why we need urlencode in generic HTTP controller?

kr0815, 2017-12-10:
I tried this afternoon to check why espeasy can´t log directly to my volkszaheler account hosted on a webhoster
I came to the point it´s because of URL encoding, originally i send 
`volkszaehler.org/htdocs/middleware.php/data/579f6430-d82b-11e7-9b16%2d319ab8bfd2e1.json?operation=add&value=19.3
`
But ESPEasy sends
`volkszaehler.org/htdocs/middleware.php/data/579f6430%2dd82b%2d11e7%2d9b16%2d319ab8bfd2e1.json?operation=add&value=19.3
`

I think my server / webhoster (1&1) can´t handle this

Question: Do we really need this encoding?

Grovkillen, 2017-12-10:
According to wikipedia "-" is not restricted but "=", "&", "?" and "/" are so I don't understand why ESP Easy would send that?

https://en.wikipedia.org/wiki/Percent-encoding

TD-er, 2017-12-10:
In WebServer.ino:

```
String URLEncode(const char* msg)
{
  const char *hex = "0123456789abcdef";
  String encodedMsg = "";

  while (*msg != '\0') {
    if ( ('a' <= *msg && *msg <= 'z')
         || ('A' <= *msg && *msg <= 'Z')
         || ('0' <= *msg && *msg <= '9') ) {
      encodedMsg += *msg;
    } else {
      encodedMsg += '%';
      encodedMsg += hex[*msg >> 4];
      encodedMsg += hex[*msg & 15];
    }
    msg++;
  }
  return encodedMsg;
}
```

It simply allows for "a..zA..Z0..9" and the rest will be marked HEX encoded.
I think there are other characters allowed, which will now be escaped.
Like "-", "_" and the characters you also mentioned, which may occur in URLs.

Grovkillen, 2017-12-10:
Yes, we need to fix that.

TD-er, 2017-12-10:
Already working on that.
My proposal is to only add the "RFC 3986 section 2.3 Unreserved Characters (January 2005)" characters

Like this:
```
    if ( ('a' <= *msg && *msg <= 'z')
         || ('A' <= *msg && *msg <= 'Z')
         || ('0' <= *msg && *msg <= '9')
         || ('-' == *msg) || ('_' == *msg)
         || ('.' == *msg) || ('~' == *msg)
```

TD-er, 2017-12-10:
Hmm now that I think of it, maybe #610 is also related to this?

TD-er, 2017-12-10:
That took way longer than needed, but I had some issues probably with encoding. The first commit appeared like the entire file had changed.

kr0815, 2017-12-17:
i tried your solution, it works for volkszaehler / 1&1 webhoster now, thanks a lot

closed 2017-12-18 by psy0rz