letscontrolit/ESPEasy #637: ESPEasy32 can't connect to hidden AP

Budman1758, 2017-12-12:
Using the latest (I assume) version of ESPEasy32 after flashing the firmware multiple times it would not connect to my router no matter how I tried to set it up.  From the web page setup page and via serial port it would not connect.  I finally went into my router settings, unhid the SSID and the module connected immediately.

After successful connection went back to router, set it for hidden SSID and rebooted the module and again would NOT connect.

Seems like ESPEasy32 does not like hidden WiFi.

Firmware downloaded from this link.
[http://www.letscontrolit.com/downloads/ESPEasy32_R20000.zip](url)

psy0rz, 2017-12-18:
i'm not sure if the normal ESPEasy is even able to connect to hidden SSIDs? Didnt test it myself yet.

Maybe someone can verify this?

Budman1758, 2017-12-19:
I believe the normal ESPEasy has no problem. I have quite a few of them running and did not notice this issue with any of them.  I will see about that just to make sure tho......

juliokele, 2020-11-13:
Connection fail if channel above 10 and SSID is hidden.
[Hidden SSID Bug](https://www.letscontrolit.com/wiki/index.php/Wifi)

Any news on this?

TD-er, 2020-11-13:
> Connection fail if channel above 10 and SSID is hidden.
> [Hidden SSID Bug](https://www.letscontrolit.com/wiki/index.php/Wifi)
> 
> Any news on this?

Hidden SSID should work (tested it myself a while ago).

Limits regarding channel > 10, sounds like a regional setting.
What region setting is present on your access point?
Have you built ESPEasy yourself or is it a nightly build?
What country do you live?
Have you tested it with other brand access point also?

juliokele, 2020-11-13:
"Limits regarding channel > 10, sounds like a regional setting."
No, other Devices working fine on e.g.: CH13

"What region setting is present on your access point?"
Germany

"Have you built ESPEasy yourself or is it a nightly build?"
Release mega-20201102 from githubESP_Easy_mega_20201022_test_ESP32_4M316k

"What country do you live?"
Germany

"Have you tested it with other brand access point also?"
No, only on [Unify UAP-AC-LR ](https://www.ui.com/unifi/unifi-ap-ac-lr)

TD-er, 2020-11-13:
Channel 11 should be available all over the world, so it is rather strange you can't use channel 11.
Channel 12 and 13 may be problematic when using some regional settings (can also be set in the firmware)

juliokele, 2020-11-13:
Channel 11 works now, Channel 12 and 13 don't.
"can also be set in the firmware" How?

TD-er, 2020-11-13:
> Channel 11 works now, Channel 12 and 13 don't.
> "can also be set in the firmware" How?

I know there are functions in the firmware to set regional options.
As far as I know, the defaults were always set to work anywhere (as long as you don't use it as an access point on channel 12 or up)
But maybe they changed the defaults here, meaning we now have to set the regional options ourselves.
This is not yet included in ESPEasy.

What I meant to say was that it could be a (default) setting in the firmware/SDK, so it can still be a problem on the ESP side in the software -> not the hardware of the ESP.