letscontrolit/ESPEasy #640: PCF8574 Keypad plugin behavior questions.

Budman1758, 2017-12-12:
I'm attempting to create a keypad module for an outside garage door opener setup.  I am using this keypad  [https://www.adafruit.com/product/419](url) with the testing version of the PCF8574 keypad plugin.  It is working in the respect that it recognizes the the inputs from the keypad but the issue is that for every keypress you have to wait for it to process before a new input will be captured.  This translates to about 15 to 20 seconds to input a 5 digit code.

Keypad presses are not recognized at all if the module is "busy" doing ANYTHING at all.  If its processing anything, outputting anything to the serial port then no keypresses are captured.  This is kind of a show stopper from a practical standpoint.

I cant find any documentation on this setup at all so I am kinda shooting from the hip here.  I have all the rules worked out for recognizing a certain code and doing something with it.

Is this normal?
Is there a better way of wiring it?  Keypad pins are going direct to PCF8574 pins.

I'm thinking the keypad needs to interrupt the processor for a bit while a code is entered and resume when a certain key like * or # is pressed.  Is this possible?

Dev 12 "Test" firmware version BTW.

TD-er, 2017-12-12:
Yes, it is possible to mark pins to trigger an interrupt.
However, that has some impact on the iRAM (cache) memory usage.
And it will also interrupt (pun intended) other tasks, which may lead to crashing plugins because such behavior was never tested.

If you want to see an example in code to how to set an interrupt to a pin, see this pull request of my attempt to reduce cache (iRAM) usage of the software serial port: #631 
See this file: https://github.com/letscontrolit/ESPEasy/pull/631/files#diff-e1c00b8efaa00fbd65a0d47afeb3948d

Budman1758, 2017-12-12:
I don't have any code skills at all.  ESPEasy rules are about as complicated as I will ever begin to understand. :)

TD-er, 2017-12-12:
I was just looking into the chip you used. (stated in the subject line)
Is this chip perhaps a better choice? http://www.ti.com/lit/ds/symlink/tca8418e.pdf
That one also has a buffer to store entered keypresses and it could be read several times a second.
Not sure if it is supported, but that should not be too hard, judging the datasheet.

Budman1758, 2017-12-12:
That looks a bit more like something made for the task.  Reason I'm using the PCF8574 is cause its already in the firmware.  Feel like writing a plugin?  I'll buy the chips n salsa.

Budman1758, 2017-12-12:
Looking at the datasheet now it would be pretty difficult to work with that particular package on a breadboard.

TD-er, 2017-12-13:
Indeed, Ball-grid-arrays are not very useful on a breadboard :)

It was the first hit when looking for some kind of I2C keyboard controller.
The thing is that it sounds like a bad idea to poll key strokes on a matrix at such a slow interval.
There are controllers to do the real-time job and then let the ESP just do the interpretation and communications.

I just saw this setup (in Dutch) with an Arduino and the chip + keypad you already have: http://demo.jezietheternietaanaf.nl/keypad-aansluiten-via-i2c/
Perhaps you have also an Arduino laying around, just to test how responsive it can be. The ESP however isn't that responsive, because lots of other tasks have to be performed.

It might be possible to run such a plugin at a faster interval, but still it may not be as fast as the Arduino solution which doesn't have to do anything else. And I don't know what effect it may have on the rest of the operations.

I will try to order these parts, so in a month or so, when the parts arrived, I may have a look at the plugin :)

uzi18, 2017-12-13:
> keypad-aansluiten-via-i2c/

You can use small arduino nano directly connected to keypad to scan and
send keycode,  and espeasy connected to this nano board by rs232 or mimic
one of these i2c chip.

Budman1758, 2017-12-13:
I understand there are all kinds of other ways to do this but it kinda takes a lot of the "easy" out of ESPEasy.  I am not a programmer at all.  I just bash stuff together and ESPEasy takes the programming pretty much out of the equation for people like me.

Not trying to be an ass here but keypad "support" would seem to indicate you can actually use it as such.  Having to wait several seconds between keypresses doesn't really add up to "support" IMHO.

TD-er, 2017-12-13:
Like I said, I will order the parts myself and see what can be changed.
Only thing is, it probably won't be fixed before the 2.0.0 release, which is almost completed and I think I am already driving @psy0rz and @Grovkillen close to madness with all the pull-requests to be merged :)

I totally agree that entering keys like you describe is not how it should be.
Let's hope it can be fixed in software, but I expect proper key-press detection can only be done by sampling the keypad several tens of times a second and I am not really sure the combination of ESPeasy and that chip are up for that task.
But time will tell :)

uzi18, 2017-12-13:
@Grovkillen : do we have any info how long does it take to get scancode?

TD-er, 2017-12-13:
@uzi18 I just browsed through the datasheet and at first glance there is not much to say about the maximum sample rate.

But, I just saw this:

> The PCF8574 device provides an open-drain output (INT) that can be connected to the interrupt input of a microcontroller. An interrupt is generated by any rising or falling edge of the port inputs in the input mode.

This means, it is possible to connect the INT line to the controller to trigger a read.
So you have to set 4 rows to high and when any of the input lines change (by a keypress) the INT line will trigger.
Then you have to disable the trigger  and perform up to 4 samples to detect which key was pressed. Each sample consists of setting 1 output line and a read.
So I guess you have to perform about 8 read/write actions.
Each of 3(??) bytes. So that's 8*24 bit + some delay between commands. With 100kHz I2C is that less than 500 times per seconds. How much less depends on the minimum delay between writing and reading.
And often keypresses are not immediate, but bounce. So perhaps you'll have to scan for a few tens of milliseconds.

If no other devices trigger interrupts and no other I2C devices are active and no processing time is spent on other tasks like the WiFi, then it should be possible to detect the key presses accurately. But if those other interrupts are present during a keypress, the moment to scan which key was pressed may be too short to detect properly.

uzi18, 2017-12-13:
I mean if scancode takes too long we can scan one line every 20ms
80ms is not too long and with debounce 160 is enough.

TD-er, 2017-12-13:
I was just thinking, you can determine the scancode in 2 samples.

preparation: Set the rows active as high output  + activate the INT mode

INT triggers interrupt => Read inputs.
Then you know which column is active.

Next: Set that column as output and the row lines as input and set the output high.
Then read the inputs and there should be 1 row active. 
Repeat this a number of times to prevent false detection by bouncing keys.

I think the problem is not in scanning every N msec. The problem is more that there is not much of a guarantee it can be done within N msec after the INT line is triggered.
And then keypresses are easy to miss.

uzi18, 2017-12-13:
>50ms is ok for debouncing
I'm thinking about 4 samples - every row * 20ms.
Tu be sure what is going on we need to measure time to go into scan code and scanning time.
So we will now exactly how long scanning takes and how often it is launched.

TD-er, 2017-12-13:
Problem is I still have to order the parts to be able to test :)

uzi18, 2017-12-13:
You can use i2c->lcd (usualy pcf8574 based) converter and connect keypad to it or just shortcut pins there :)

Budman1758, 2017-12-13:
The thing I see in the serial output is that the keypress itself is not "seen" by ESPEasy when it is doing ANY other processing.  Is there a way to "pause" the processor until the code is done being entered?  Perhaps a # key followed by whatever code and then a * to tell the processor to resume?  In other words # tells the processor to listen up and # says OK run with it? Or maybe make use of the timers function in the rules engine?  Just throwing stuff out there to see if anything sticks.  :)

TD-er, 2017-12-13:
I just looked at the P61 keypad plugin.
It looks like the INT line is not being used in the plugin.
I guess it can only work properly if either the keypad is scanned continuously (won't happen on the ESP8266) or when an interrupt is triggered.
So to get it to work properly, the INT line should be connected to a GPIO pin of the ESP and that pin should be set to call a function when triggered.
That function should be the reading of a single key stroke and re-enabling the INT line.

Budman1758, 2017-12-13:
That sounds like something that needs to be a part of the plugin.

TD-er, 2017-12-13:
Indeed, I guess it would help a lot.

Grovkillen, 2017-12-14:
@TD-er Your work is much appreciated :smile: Don't mind the PR at all. I wish we had two reviewers at the moment. My part is only administration since my expertise in the coding is not yet high enough for any reviewing.

uzi18, 2017-12-14:
> @TD-er <https://github.com/td-er> Your work is much appreciated 😄 Don't

I realy think pooling is enough for matrix keyboard.

TD-er, 2017-12-15:
@uzi18 It should be enough, but I am afraid the ESPeasy architecture isn't that "real-time" enough for sampling the keypad matrix and not to miss any key.
When adding the extra pin, you can get the focus immediately and allow for a few tens of msec to sample the key pressed and continue doing the other tasks.
It may improve reliability a lot, without running the sampling function every 1/50th of a second (shortest interval in ESPeasy)

uzi18, 2017-12-15:
Is it possible to build espeasy without platformio?

I want to add time measure and check if reading keyboard is accurate

TD-er, 2017-12-15:
Yes, you can also build it using the Arduino "IDE".
Make sure you select the correct upload mode.
And if you checkout another branch via Git, you probably want to reload the Arduino IDE, since it doesn't detect filechanges very well.

Budman1758, 2017-12-16:
Just in case anybody is interested I purchased a HT16K33 keypad/matrix driver chip and set it up and it behaves the same. Still have to wait a few seconds between keypresses so that they are not missed. This chip is designed for this sort of thing so I thought perhaps it would behave differently.  Should this be a separate issue?

psy0rz, 2017-12-18:
It seems P061_keypad is already scanning 50 times per second. @Budman1758 is it possible to change something on your controller so it responds faster? Which controller you use?

Also, perhaps even more important: Did you set the MessageDelay setting to 0? Otherwise ESPEasy will deliberately wait at least that many seconds between two events.

Budman1758, 2017-12-19:
Not using any controller like Domoticz or anything.
Message delay in the keypad device page is at zero.  Keypress seem to respond immediately regardless of the delay setting.

When there are no rules to process for the keypresses they register just about as fast as you can punch them in.  The delay is caused by the rules processing. As they are being processed is when they are being missed.  As soon as you punch in 1 key of say a 4 key code its processing that first key and misses the next ones during processing. This literally makes it take 10 to 15 seconds to enter 4 numbers and an "enter" key like # or *. There is also a short wait for the plugin (I assume) to report a zero scancode after each keypress.  For each keypress the serial output reports 4 events like this......

Mkey : key=0x11
EVENT: keypad#ScanCode=17.00
((if you have rules that processing shows here. If no rules nothing here))
Mkey : key=0x0
EVENT: keypad#ScanCode=0.00

Until that last line shows up in the serial output it will not process another keypress.

Is there any way to set it up to wait for a certain key b4 processing? Much like when you enter your pin at an atm. Nothing happens until you hit "enter"  I understand that it would take some major code magic for that to happen. I'm afraid that its not very user friendly as it is now. A short wait AFTER you hit "enter" is perfectly acceptable. Having to wait a few seconds between numbers is a bit frustrating because unless you have a visual indication that it "got it" you don't know if it "got it" or not.

psy0rz, 2017-12-19:
So what kind of rules you have that makes it so slow?

uzi18, 2017-12-19:
Also please provide more log lines

Grovkillen, 2017-12-19:
And embed them using the code formation for readability. 

```
Like this
And this...
```

Budman1758, 2017-12-19:
Ok.  I did a bit of optimizing of the rules and its a bit faster. Still must wait about 2 seconds between presses tho...  Rules explained.
Each keypress creates an event.
0-9 creates "number" event
"*" key creates "Send" event  ( =to "enter") 
"#" key creates "clear" event
"Number" event creates 2 dummy values and does the math to create a unique value for each number entry.  This math was suggested by @Grovkillen. It has the effect of creating a unique value even if you enter the same numbers in a different order. So a particular 4 numbers add different for any sequence used.
"Send" event compares dummy value and if certain value sends command to another unit to make something happen.
"Clear" sets dummy values to zero for the next go round.

I don't  why but everytime I paste the rules into the insert code function in this post all I get is regular text. Rules.txt attached.
[Rulesupload.txt](https://github.com/letscontrolit/ESPEasy/files/1573470/Rulesupload.txt)

Some log output. (level 4) Log.txt attached.
[log.txt](https://github.com/letscontrolit/ESPEasy/files/1573494/log.txt)

I also have a crashing problem every so often.  You can see that in the log file.

Side note... How do you get that nice code format text in this window? I use the insert code function at the top of the editor and all I get is the same plain text as all the other text in the message. What am I doing wrong?  :(

uzi18, 2017-12-19:
EVENT: clear
EVENT: Error: Nesting level exceeded!
EVENT: Error: Nesting level exceeded!
EVENT: Processing time:81 milliSeconds
EVENT: Processing time:4163 milliSeconds

Budman1758, 2017-12-19:
I saw that.  Where?

uzi18, 2017-12-19:
```
EVENT: Processing time:134 milliSeconds
KPad : ScanCode=0x0
EVENT: pcf#ScanCode=0.00
EVENT: Processing time:135 milliSeconds

KPad : ScanCode=0x14
EVENT: pcf#ScanCode=20.00

EVENT: Processing time:136 milliSeconds
KPad : ScanCode=0x13
EVENT: pcf#ScanCode=19.00
EVENT: Processing time:135 milliSeconds
KPad : ScanCode=0x12
EVENT: pcf#ScanCode=18.00
EVENT: Processing time:135 milliSeconds
KPad : ScanCode=0x11
EVENT: pcf#ScanCode=17.00

EVENT: Processing time:134 milliSeconds
KPad : ScanCode=0x0
EVENT: pcf#ScanCode=0.00
EVENT: Processing time:135 milliSeconds

```

Budman1758, 2017-12-19:
OK????

uzi18, 2017-12-19:
For me it is ok, now i will test it with some of Your rules, but without pulse ...

Budman1758, 2017-12-20:
As I have said before, It responds pretty fast if its not having to actually do anything after the keypress. The short pulse of the GPIO is to flash an led to show that the keypress was actually registered.

uzi18, 2017-12-20:
So use timers instead.
Scancode takes 300-450 ms 

my 1st task is keypad 2nd is dummy
```

On System#Boot Do
  Event,Clear
Endon

On keypad#ScanCode Do
  IF [keypad#ScanCode]=17
    Event,Number=1
  Endif
  IF [keypad#ScanCode]=18
    Event,Number=2
  Endif
  IF [keypad#ScanCode]=20
    Event,Send
  Endif
  IF [keypad#ScanCode]=19
    Event,Clear
  Endif
endon

On Clear Do
  TaskValueSet,2,1,0
  TaskValueSet,2,2,0
Endon

On Send Do
  TaskValueSet,2,1,0
  TaskValueSet,2,2,0
Endon

On Number Do
  TaskValueSet,2,1,[temp#a]+%eventvalue%*10^[temp#b]
  TaskValueSet,2,2,[temp#b]+1
Endon

```

uzi18, 2017-12-20:
So use timers instead.
In number event just set gpio and start timer, add event to reset gpio after timer elapsed

Budman1758, 2017-12-20:
How does that speed up the processing of the scancode?

uzi18, 2017-12-20:
Looks like Pulse cmd blocks data processing.
But to be sure i need to look into pulse code.
Try to use timer

Budman1758, 2017-12-20:
It does block.  I shortened the pulse to 200 and its that much faster now. Still must wait 2 to 3 seconds between keypresses for them to all register. If a timer setup can pause the "number crunching" that would be great. That would have to be done b4 the "send" event processes though. "Timing therefore probably gets a bit complicated.

Budman1758, 2017-12-20:
All I'm trying to do is make it more natural like any other "pin pad". Its not natural to pause between entering each number.

uzi18, 2017-12-20:
First remove or comment out pulse cmds
Test responsiveness
Try to add timer to work for your led notify
Test one more time

Budman1758, 2017-12-20:
Hard to actually measure but its a tad bit more responsive. Still must pause about 2 or 3 seconds between keypresses.

Grovkillen, 2017-12-20:
Why do you add '.3' to the tenth? This line...

```
TaskValueSet,1,2,[Dummy#TempTenth]+.3
```

Should be...

```
TaskValueSet,1,2,[Dummy#TempTenth]+3
```

In my opinion.

Budman1758, 2017-12-20:
I don't know.  I used the example you gave me in your forum post. Maybe I just added a period there by mistake. Does it make any real difference? Would it "speed up" the number crunching?

uzi18, 2017-12-20:
Look into my rules,  there is +1,  dont investigate something unnecessary,
just use it as it works ok.

Grovkillen, 2017-12-20:
@uzi18 yes that was my first proposal as well but @Budman1758 wanted to use the NN numbers without conversion into "real" 0...9 numbers. That is why I suggested `+3`. :smile: 

But my suggestion is to try remove all plugins that are not needed, recompile a new binary and test. Maybe even cut down on possible tasks for the loop to be as short as possible.

uzi18, 2017-12-20:
for testing it could be +1, after resolve issue it coult be +3

@Grovkillen I have flashed dev-dev12 + clean config, added dummy and keypad plugin and it is work for me, so no recompilation needed.

Grovkillen, 2017-12-20:
@uzi18 no delays at all? rules work as intended etc. etc.?

uzi18, 2017-12-20:
Yes, but I didn't tested with pulse command in rules

Grovkillen, 2017-12-20:
The pulse was only used to indicate the input with a led I suspect? I think
we should then close this issue since you could make it work? Maybe use a
timer as you suggested, and turn it on on first key press and off on
"enter/send pin"?

Budman1758, 2017-12-20:
The 100 ms pulse was there to show you that each keypress was "seen" Starting a timer and turning it off later completely defeats its purpose. That pulse added 1 tenth of 1 second to the delay. Not much at all. I have always said that this works. The point is that you still have to consciously pause about 2 seconds between keypresses. That is not intuitive or natural. Time yourself entering a 4 digit pin plus enter and I venture a guess that you can and DO get it done in about 3 or 4 seconds. That does not work in this setup.

I have no problem with the way it works for my application. My only point here is if you were to try and market this it would probably not do very well. I'm not suggesting that I am going to do that at all, just that people in general would not be happy with it.

My question about waiting to process the entered numbers until _after_  you hit a defined "enter" key was never addressed or answered. Is or is that not possible? If not so be it. If it is that would most likely solve this entire issue.

The only other thing here is my setup is crashing on a regular basis now.  I posted a screen dump up above in this thread. I can do it again if necessary. Or should that be a separate issue?

Grovkillen, 2017-12-20:
But please verify if removing the pulse command make the unit responsive.

Also:
Given the "eventvalue=N" where N is 0...9 and use the rule Uzi gave you,
you'll have the PIN in real value except backwards.

If you want a fixed number of digits in the PIN we can provide you a rule
for that (making the PIN display correctly, i.e. not backwards).

uzi18, 2017-12-20:
If You put like this it will be natural:
TaskValueSet,2,1,[temp#a]*10^[temp#b]+%eventvalue%

Grovkillen, 2017-12-20:
@uzi18 dooh! I made it over complicated 👶

Budman1758, 2017-12-20:
It is _very slightly_ more responsive. But by the same token, you tend to input the numbers faster and end up where we started. The idea of the led flash is you then know the number was received.

I don't really care what unique number we end up with. This method does not require any particular amount of numbers in the "pin" number. All you do is pick a code, see what it adds up to and adjust the rule to reflect that particular number.

uzi18, 2017-12-20:
Did You tried to use timer for led?

[28 more replies]

closed 2018-03-16 by Grovkillen