letscontrolit/ESPEasy #641: Two RFID Wiegand

SamNitro, 2017-12-16:
Hello,
I have same problem with two RFID Wiegand Reader on one Nodemcu.
https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=3655&p=21143&hilit=wiegand#p19946

Have anyone a solution?
Thanks

Grovkillen, 2018-10-23:
Is this still an issue? If so please reopen.

rwytrwal, 2019-02-21:
Why its closed when its still a bug?

trynifred, 2019-03-11:
Bonjour, Y a-t-il un avancée ? J'ai le même problème et aurait vraiment besoin d'une solution.
Merci

rwytrwal, 2019-11-06:
is there any progress since 2017?

josh-79, 2022-10-22:
This problem still exists in version "mega-20220809_57b0c0c". I'm using two wiegand readers. No matter at which reader I read the card, I always get the message from the first Wiegand reader via MQTT. This makes it impossible to identify which reader was used. Unfortunately, this is very bad with multiple doors.

tonhuisman, 2022-10-22:
Can you show your ESPEasy MQTT Controller configuration?

josh-79, 2022-10-22:
Hi, thanks for the quick reply. Here are the screenshots:

[image]
[image]
[image]
[image]
[image]
[image]

tonhuisman, 2022-10-22:
And what ESPEasy bin file you have installed?

josh-79, 2022-10-22:
Right this is the installed firmware:

[image]

tonhuisman, 2022-12-27:
@josh-79 Can you upgrade to the latest ESPEasy [release 20221224](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20221224), as at the time of your issue report, the Wiegand plugin _didn't yet_ support multiple instances of the same plugin on a single device (though the code changes supporting that was already merged, but it wasn't yet released).
Most likely this has been resolved (I own only 1 Wiegand device, so can't do simultaneous tests).
Please keep us updated with your findings.

tonhuisman, 2023-09-01:
This has been solved, as the related PRs have been merged for some time.

closed 2023-09-01 by TD-er