letscontrolit/ESPEasy #642: enable "send to controller" multi-sensors seperatly

kr0815, 2017-12-17:
Wouldn´t it make sense to activate the indivudual channels of a multi-sensor seperatly?
I have a BMW280 for example, need pressure and humidity, the temperature is not needed, as it´s even an internal value to correct pressure,

At the moment i could only enable or disable all channels together to send their values to my controller

TD-er, 2017-12-17:
I know a lot of Bosch sensors, but not (yet) the BMW280.

But apart from that, it sounds like a good suggestion, to allow separation of the values from a sensor.
And perhaps combining could also be an option that has a few use cases.

kr0815, 2017-12-17:
sorry, ma fault, bme280, but this should apply to all sensor with more than one value

Grovkillen, 2017-12-17:
Maybe an enable tick for each value?

TD-er, 2017-12-17:
Or a separate option (plugin?) to format the output, just like in the display plugins, where you can define the output of each sensor.

Grovkillen, 2017-12-17:
Ah, that's pretty clever.

kr0815, 2017-12-17:
Mabe easier to have a enable tick at the position where also formula and decimals could be entered?
Or, as a workaround, formula could be set to something like "ignore" ?

TD-er, 2017-12-18:
The more I think about it, I think it will be a good idea to add a "Composite" device, which will output to a specific controller (selectable like any other plugin) which will allow to format the desired output with a selectable set of values from any other device present.
This will definitely be a 2.1 feature (or later), but I think I will start some test version later this week, just to get a feeling about it. (like taking a note in C++ :) )

kr0815, 2017-12-18:
If i undertsand you right, you think about a device that collects all desired values from different controllers and puts them in one packet?
I have to confess, i don´t really like the idea, because many controllers, as for example my volkszaehler, need the value to be send individually for each channel

TD-er, 2017-12-18:
"Device" has to be seen as a virtual device.
And it should not disable the present feature to upload values.
Just an extra "device" which is able to collect data from more than one sensor and send it in the order one wants.
And ofcourse it could also send just one value per virtual device.
You should see it as a flexible way to allow for special wishes, which could allow to interact with all existing plugins, instead of changing plugins for just a special use case.

Volkszaehler may want values to be sent one-by-one, but Domoticz wants to have values combined which are from the same source (room/set of sensors)
Another thing is that sending the battery status, which Domoticz supports, is not yet possible since that data should be taken from another sensor and currently it is not possible to combine sensors.

Only thing is that you then definitely want more than 12 devices possible in ESPeasy.

TD-er, 2017-12-19:
This is related to issue #232 and #564 
If implemented like discussed above, these issues will also be fixed.