letscontrolit/ESPEasy #643: Bug in rules save location in flash?

Budman1758, 2017-12-18:
I have seen this now on 3 separate modules now.  All of them are -07 chips.  When booting the chip with empty rules (or no rule files) everything is normal like so.....

INIT : Booting version: v2.0.0-dev12
INIT : Warm boot #8
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
INIT : Free RAM:23360
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be Keypad_Test_21 with address 192.168.4.1
WIFI : Connecting xxxxxxxxxx attempt #1
WIFI : Connected! IP: 192.168.1.184 (Keypad_Test_21)
EVENT: System#Boot
WD   : Uptime 0 ConnectFailures 0 FreeMem 19920

However, when there are some rules I get this when booting.....

INIT : Booting version: v2.0.0-dev12
INIT : Warm boot #9
FS   : Mounting...
FS   : Mount successful, used 77057 bytes of 957314
INIT : Free RAM:23360
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be Keypad_Test_21 with address 192.168.4.1
WIFI : Connecting xxxxxxxxx attempt #1
WIFI : Connected! IP: 192.168.1.184 (Keypad_Test_21)
EVENT: System#Boot
ACT  : Event,Clear
EVENT: clear
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
ACT  : TaskValueSet,1,1,0
ACT  : TaskValueSet,1,2,0
ACT  : pulse,14,1,500
SW   : GPIO 14 Pulsed for 500 mS
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
FS   : Error while reading/writing config.dat in 926
WD   : Uptime 0 ConnectFailures 0 FreeMem 19944

This is consistent and repeatable.  If I remove the rule files or just empty and save the errors disappear. Testing version of DEV `12```  I thought it was just a bad module but this is the third one and it is the first time I flashed it after unwrapping it.  Please let me know what other info is needed.

Grovkillen, 2017-12-18:
#588

This is fixed but not released.

closed 2017-12-18 by Grovkillen

psy0rz, 2017-12-18:
will be fixed in dev13