letscontrolit/ESPEasy #646: IDX/VAR missing for Generic HTTP controller

paulv888, 2017-12-19:
### Steps to reproduce
Setup a generic HTTP controller
Setup any device, and notice the IDX is not available next to "Send to Controller"

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
IDX should be available and its value should be send as %id% in the HTTP Get
This is the behavior in R120. https://www.letscontrolit.com/wiki/index.php/EasyProtocols#Generic_HTTP 

### Actual behavior
A "0" is send when using %id% in the controller Publish
i.e. esp.php?device=%id%&param=%valname%&value=%value% yields
?device=0&param=Voltage&value=5.02

### System configuration
Hardware:
ESP-12E

Software or git version:
Build: | 20000  - Mega
GIT version: | v2.0.0-dev12
Plugin sets: | [Normal] [Testing] [Development]

[image]

Grovkillen, 2017-12-19:
I should be called "ID" I guess, since IDX is a Domoticz term.

psy0rz, 2017-12-28:
@paulv888 can you make a PR if your change seems to work?

paulv888, 2017-12-29:
Done

commit 493bfd56 (2018-01-02): usesID = true (#666)

fixes #646
closed 2018-01-02 by psy0rz