letscontrolit/ESPEasy #647: Feature request: Security

filippoladerchi, 2017-12-19:
Would it be possible to add to the http commands ( eg: http://<espeasyip>/control?cmd=<command>) to only be accepted from specific mac addresses or ip addresses.

TD-er, 2017-12-19:
Sounds like a good feature.
Maybe also for the webinterface?

Do you also have any idea about how to make the unit accessible again when you locked-out yourself?
I was thinking about not having this filter active within N minutes after boot?
And I guess just having an IP-subnet as allowed would be sufficient? (e.g. 192.168.1.0/27 for 192.168.1.0 ... .31)

First we have to look into the HTTP-request, to see whether the originator is present in the request.

Grovkillen, 2017-12-19:
I believe that v2.1.0 will be focusing on security.

filippoladerchi, 2017-12-19:
When security is turned on, you will have to put at least 2 mac or IP addresses.

Grovkillen, 2017-12-19:
Since it's possible to add a typo to this list of mac/ip addresses a backup restore might be needed? Or should a complete wipe of flash be the only option out? (Not against any, just added my first thought).

TD-er, 2017-12-23:
I just added a pull request for the Mega branch (2.1.0) to add some client IP filtering.

Please read the description along pull request #653 to get some idea of this filtering.
The active IP filter settings are also displayed in the advanced info page.

filippoladerchi, 2017-12-27:
I currently make security with a mac address through my router (so if ip address changed in my pc or laptop  does not care)

ShardanX, 2017-12-27:
Mac filtering isn't a real security. Mac addresses can be sniffed from the net and spoofed easily.
Some network cards have a settable Mac. See https://www.howtogeek.com/192173/how-and-why-to-change-your-mac-address-on-windows-linux-and-mac/

papperone, 2017-12-27:
as well all Arduino ETH shields can be configured with whatever MAC address you like :)

TD-er, 2017-12-27:
@filippoladerchi 
I created a test build, based on the Mega branch for you to test: https://www.dropbox.com/s/v2q4blv764s9atd/mega_test_Security_issue647.rar?dl=0
It also includes the WiFi fixes when WiFi is unreachable and the bugfix in PubSubClient (MQTT controller using hostname)
Hopefully, I did not miss any needed commits to make this build.
So testing it would be really nice.

filippoladerchi, 2017-12-29:
Hi
The build work and I also put a different subnet than it is esp and work

closed 2018-01-02 by psy0rz