letscontrolit/ESPEasy #650: ESP-01 (1MB) flashing problems

faisalthaheem, 2017-12-21:
### Steps to reproduce
Flash a fresh esp-01 module (black) with any 1MB bin

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
yes

### Expected behavior
Tell us what should happen?
Should start normally

### Actual behavior
Tell us what happens instead?
board keeps resetting and complaining about the same problem. I have tried flashing with different baud rates and modes, but nothing works. The same firmware works great on wemos d1r2.

### System configuration
Hardware:ESP-01 1MB version
Software or git version: v2.0.0-dev12

here''s the output from the serial interface

RESET: Warm boot, reset count: 1
RESET: formatting...
RESET: formatting done...
FS   : Error while reading/writing config.dat in 878
FS   : Error while reading/writing security.dat in 878
FS   : Error while reading/writing notification.dat in 878
FS   : Error while reading/writing config.dat in 922
FS   : Error while reading/writing config.dat in 895
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
⸮U

INIT : Booting version: v2.0.0-dev12
INIT : Warm boot #2
FS   : Mounting...
FS   : Mount successful, used 1255 bytes of 113201
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 2
RESET: formatting...
RESET: formatting done...
FS   : Error while reading/writing config.dat in 878
FS   : Error while reading/writing security.dat in 878
FS   : Error while reading/writing notification.dat in 878
FS   : Error while reading/writing config.dat in 922
FS   : Error while reading/writing config.dat in 895
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
⸮U

INIT : Booting version: v2.0.0-dev12
INIT : Warm boot #3
FS   : Mounting...
FS   : Mount successful, used 1255 bytes of 113201
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 3
RESET: Too many resets, protecting your flash memory (powercycle to solve this)
FS   : Error while reading/writing config.dat in 922

PID:0
Version:0
INIT : Incorrect PID or version!
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 3
RESET: Too many resets, protecting your flash memory (powercycle to solve this)

Exception (0):
epc1=0x40106f19 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff3500 end: 3fff3770 offset: 01a0

>>>stack>>>
3fff36a0:  0000001c 00000000 3fff2680 401004d8  
3fff36b0:  3fff0ab0 0000004e 3fff4f9c 40205cb9  
3fff36c0:  00000000 00000001 00000119 4010020c  
3fff36d0:  0000001c 00000000 3fff2680 00000000  
3fff36e0:  0000001c 00000000 3fff2680 40201710  
3fff36f0:  00000001 3fff37c8 00000000 40203d13  
3fff3700:  3fffdad0 3fff2680 3fff4f9c 3fff2740  
3fff3710:  3fffdad0 3fff2680 3fff1464 402142f1  
3fff3720:  3fffdad0 3fff2680 3fff1464 402319b9  
3fff3730:  00000000 00000000 00000000 3fff4fac  
3fff3740:  0000002f 00000013 feefeffe feefeffe  
3fff3750:  3fffdad0 00000000 3fff273a 402033f4  
3fff3760:  feefeffe feefeffe 3fff2750 40205e54  
<<<stack<<<

TD-er, 2017-12-21:
For another issue-thread I created these test builds yesterday: https://www.dropbox.com/s/tr2w99jw6hfgdph/ESPeasy_issue640_test_20171220.rar?dl=0
Could you please try one of these test builds?
Just to test if it is a recently fixed bug, which is not yet released in a build dev-13.

TD-er, 2017-12-22:
Just to verify it is not a broken ESP device, can you also test another version, like the 'old' v120: http://www.letscontrolit.com/downloads/ESPEasy_R120.zip
Or just as a test, perhaps Tasmota?

It kind of looks like you cannot write to parts of the flash.
Are you certain it is a 1MB flash version and not perhaps a 512 kB version?

faisalthaheem, 2017-12-22:
I have tried flashing "ESPEasy_R120_1024" and it works, however, the tasks do not function correctly, for example, I am unable to read DHT-11.

I have also tried flashing sketches that perform various functions (DHT, MQTT etc) and they appear to be working normally, however when I flash any recent version of easyesp (ESPEasy_R147_RC8 and upwards) I run into the same restart issues.

I'm certain it's a 1MB version (the table mentions below with ** that amount of flash size may vary with board models), I have verified the same using esptool, output follows

flash_id output
==========
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 6014
Detected flash size: 1MB
Hard resetting...

chip_id output
===========
esptool.py v2.2
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x004aae02
Hard resetting...

psy0rz, 2017-12-28:
which version file of 2.0.0-dev12 did you flash?

funtastic99, 2018-01-04:
Hi,
i have the same problem. It seems that the newer ESP-01 are delivered with a different flash chip.

The working ESP-01 report as flash_id:
Manufacturer: e0
Device: 4014

and the new ones that are not working:
Manufacturer: 85
Device: 6014

Any idea how to get the new one working ?

TD-er, 2018-01-04:
Apparently, that is the GIGADEVICE_GD25LQ80 flash chip.
It looks there have been issues in the past to support that chip: https://www.google.nl/search?q=GIGADEVICE_GD25LQ80
I am not sure about how the programming does handle different chips, but maybe the flashtool has to be upgraded to support those chips? Although it looks like these chips should already be supported for well over a year now.
Does ESP easy have to know about these chips too? And could this knowledge be located in the ESP core so we have to upgrade to 2.4.0 to get it working?

funtastic99, 2018-01-04:
ESP core 2.4.0 also cannot handle this chip yet. I also created an issue for that. Maybe we can only wait for them to analyze and support this new flash.

Pablo2048, 2018-01-04:
Can anyone with this problem try to burn with DOUT mode (https://github.com/espressif/esptool/wiki/SPI-Flash-Modes)?

funtastic99, 2018-01-04:
Hi Pablo,
I already tried that but I will try again this evening.

funtastic99, 2018-01-05:
Still does not work. Also not with DOUT. But to have this clear: everything is working fine. The firmware runs but only SPIFFS is not working with the news flash

TD-er, 2018-01-05:
So making changes to the settings is not possible?

eriktews, 2018-01-05:
How are you actually supposed to get it working for the first time. As far as I understood ESPEasy, you have to flash it and when you do the initial configuration, the firmware has to write to the flash, right?

Also in the meantime, it looks like it's hard to predict which chip you will get when you buy a ESP-01 from an online show. Does anyone have a suggestion where I can buy the old modules?

HarryHase11, 2018-01-05:
But ESPEasy_R147_1024.bin works fine.

moo2089, 2018-01-08:
hi,
same problem here,
mod type: esp8266 esp01s
flash P25Q80 8Mbit

r147, r148 ok, but above them nothing is working properly, continous watchdog resets
tried different setups, ext. psu, ftdi built in psu, 2 ftdi module, 2 esp module, 470uF cap and 100nF also added, same problem, confused

psy0rz, 2018-01-10:
strange..normally a reset should format the SPIFFS and everything should be fine.

From the logs above:
```
RESET: formatting...
RESET: formatting done...
FS : Error while reading/writing config.dat in 878
FS : Error while reading/writing security.dat in 878
FS : Error while reading/writing notification.dat in 878
FS : Error while reading/writing config.dat in 922
FS : Error while reading/writing config.dat in 895
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)
```

Those errors while writing are the cause of all the problems: If that steps fails, your ESPEasy is unusable. 

I think there where other issues with the same problems? Didn't it have to do with the flash-mode of the flash tool or something?

HarryHase11, 2018-01-10:
> but it's only with espeasy -> tasmota and R148 works fine on this device

TD-er, 2018-01-10:
Could it be that those other versions (Tasmota, Espurna, ESPeasy R148) have a sketch size < 512 kB?

psy0rz, 2018-01-10:
I think so, since mega was especially for >=1M devices only. 

if that is the problem in this case, we might need to build in a extra flash-size check that warns and stops the boot process?

TD-er, 2018-01-10:
It looks like it is a 1 MB flash. But if I remember correctly, the ESP8266EX will load up to 1 MB into memory, which means the sketch may never be more than 1 MB.
However, since I saw a lot of ESP-01 boards to be with only 512k flash and also support for OTA was sometimes different, I thought maybe the ESP01 was also limited in the max. sketch size, or perhaps it cannot access the flash like the "bigger" ones do.

HarryHase11, 2018-01-10:
can anyone build a testversion < 1MB to test with this devices?

TD-er, 2018-01-10:
I think we can disable the OLED framed module? That one is quite big with its font embedded.

HarryHase11, 2018-01-10:
Let's try; I can't compile it myself yet; can anybody do it?

psy0rz, 2018-01-10:
@HarryHase11 so your module only has 512kb flash?

HarryHase11, 2018-01-10:
i bought it with 1MB, but not sure what wrong; and that problem some other user have the same issue.

psy0rz, 2018-01-10:
ah k thx

HarryHase11, 2018-01-10:
found some hints in the internet that this flash chip have only 512KB size.

psy0rz, 2018-01-10:
can you checkout the sysinfo page? it should tall you how much flash you have.

moo2089, 2018-01-10:
hi,
this is mine, with same issues:

Load: | 8% (LC=32657)
-- | --
Uptime: | 85 minutes
Wifi RSSI: | -45 dB
IP: | 10.0.0.15
GW: | 10.0.0.12
Build: | 148
Core Version: | 2_3_0
Unit: | 2
STA MAC: | 2c:3a:e8:4a:bf:cf
AP MAC: | 2e:3a:e8:4a:bf:cf
ESP Chip ID: | 4898767
Flash Chip ID: | 1335429
Flash Size: | 1024 kB
Flash Writes (since boot): | 2
Sketch Size/Free: | 421 kB / 516 kB
Free Mem: | 26792
Devices: | 35
Boot cause: | Cold boot

psy0rz, 2018-01-10:
ok so its not flashsize

psy0rz, 2018-01-10:
does someone have a link where to buy this type?

moo2089, 2018-01-10:
wha do you mean?
module itself?

psy0rz, 2018-01-10:
yeah the module that it is giving these problems. then i'll order a few so i can test/debug.

moo2089, 2018-01-10:
https://www.ebay.com/itm/ESP8266-ESP-01S-Serial-WIFI-Wireless-Module-Adapter-Breakout-Send-Receive-AP-STA/311736436700?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

moo2089, 2018-01-10:
this is what i use in this moment

moo2089, 2018-01-10:
and i ordered some 8Mbit and 32Mbit flash to relpace, to see what happens

psy0rz, 2018-01-10:
thanks..i've ordered 5 but esptimated shipping is 30 days :(

moo2089, 2018-01-10:
i have got it within 2 weeks (Hungary)

HarryHase11, 2018-01-10:
can't you build a binary with smaller size only for testing, i thing that will work

psy0rz, 2018-01-10:
if that is the case, then all 1Mb modules should give problems at this moment, not only yours.

TD-er, 2018-01-10:
I think it is probably more ESP01 related than flash size.

moo2089, 2018-01-10:
i agree, this is very small sketch, also not working!
https://www.letscontrolit.com/wiki/index.php/ESPEasy#Loading_firmware
 
2.0+ firmware on 1MB modules
If you are running the stock Mega edition firmware on 1MB modules, you will find that you can no longer load newer images because there's insufficient free space to upload the image (344kB as of 2.0.0.-Dev 9 edition). You can compile smaller images to solve this issue or use a small image that has been build just to upload newer images. So you would always have to use a two-step process to renew using OTA on 1MB modules. Using the uploader image as an in-between step, you can use images up to 604kB.

ESPEasyUploaderMega

The uploader uses SPIFFS so it uses the same configuration regarding SSID, WPA key and IP. It will connect to your Wifi network as usual, but it will only present this small web page:

ESPUploader.png

Tutorial_ESPEasy_Firmware_Upload

TD-er, 2018-01-10:
I assume these modules were not updated OTA?
But rather via a flash tool?

moo2089, 2018-01-10:
anyway, i ordered different esp01 modules, to see if it is module related issue, or what?

moo2089, 2018-01-10:
i tried OTA and serial flash as well, with same issues

HarryHase11, 2018-01-10:
at the weekend i take a look on the esp01-flash chip which are working

TD-er, 2018-01-10:
Please try this test build:
https://www.dropbox.com/s/t98zoqeouafhkvx/512k_Firmware_issue650_v2.0_stripped.rar?dl=0

I removed:

- P016_IR
- P030_BMP280
- P036_FrameOLED
- P042_Candle
- P055_Chiming

It is based on the latest v2.0 branch.

funtastic99, 2018-01-10:
You might have the same problem than described here: https://github.com/esp8266/Arduino/issues/4061

TD-er, 2018-01-10:
Hmm, converting the ESP-chip ID, posted by Harry is: 2160580
Converted to HEX is still the last 3 bytes of the MAC address.
So still no way to detect these versions :(

HarryHase11, 2018-01-10:
TD-er: thx, test build has the sam problem, so it's not the sketch size

psy0rz, 2018-01-10:
thanks @funtastic99  , https://github.com/esp8266/Arduino/issues/4061 seems it might be the same indeed.

relax81, 2018-01-12:
I'm glad I found this thread because I'm having the same issue. When I flash the v120 everything runs fine but I always get the above errors with the v2 versions.

I've received the ESP-01 a few days ago - ordered it from Banggood
https://www.banggood.com/Upgraded-Version-1M-Flash-ESP8266-ESP-01-WIFI-Transceiver-Wireless-Module-p-979509.html

grigorygn, 2018-01-15:
It seems to me we have PUYA P25Q80H flash chip incompatibility.

I have ESP01 mmodules with BergMicro flash and PUYA as well.
I've flashed latest ESPEasy_mega-20180114 to them.
Modules with BergMicro start fine with no errors:

> BergMicro  25Q80ASSIG
> 
> Boot OK!
> 
> INIT : Booting version: mega-20180114
> INIT : Warm boot #1
> FS   : Mounting...
> FS   : Mount successful, used 75802 bytes of 113201
> INIT : Free RAM:25800
> INIT : I2C
> INIT : SPI not enabled
> WIFI : AP Mode disabled
> WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
> WIFI : Could not connect to AP!
> WIFI : AP Mode enabled
> WD   : Uptime 0 ConnectFailures 0 FreeMem 20568

Modules with PUYA have continuous bootloop:

> PUYA  P25Q80H
> 
> Bootloop...:
> 
> INIT : Booting version: mega-20180114
> INIT : Warm boot #2
> FS   : Mounting...
> FS   : Mount successful, used 2008 bytes of 113201
> FS   : Error while reading/writing config.dat in 1013
> 
> PID:0
> Version:0
> INIT : Incorrect PID or version!
> RESET: Resetting factory defaults...
> RESET: Warm boot, reset count: 1
> RESET: formatting...
> RESET: formatting done...
> FS   : Error while reading/writing config.dat in 960
> FS   : Error while reading/writing security.dat in 960
> FS   : Error while reading/writing notification.dat in 960
> FS   : Error while reading/writing config.dat in 1013
> FS   : Error while reading/writing config.dat in 983
> RESET: Succesful, rebooting. (you might need to press the reset button if you've                                                                                                                                                              justed flashed the firmware)
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1384, room 16
> tail 8
> chksum 0x2d
> csum 0x2d
> v09826c6d
> ~ld
> 

I've replaced PUYA P25Q80H flash chip by WINBOND 25Q32FVSIG (4MByte) and module starts to work:

> WINBOND  25Q32FVSIG
> 
> Image for 1MB flah chip
> Boot OK!
> 
> INIT : Booting version: mega-20180114
> INIT : Warm boot #1
> FS   : Mounting...
> FS   : Mount successful, used 75802 bytes of 113201
> INIT : Free RAM:25928
> INIT : I2C
> INIT : SPI not enabled
> WIFI : AP Mode disabled
> WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
> WIFI : Could not connect to AP!
> WIFI : AP Mode enabled
> WD   : Uptime 0 ConnectFailures 0 FreeMem 20696

> Image for 4MB flash chip
> Boot OK!
> 
> INIT : Booting version: mega-20180114
> INIT : Warm boot #5
> FS   : Mounting...
> FS   : Mount successful, used 0 bytes of 957314
> RESET: Resetting factory defaults...
> RESET: Warm boot, reset count: 1
> RESET: formatting...
> RESET: formatting done...
> FILE : Saved config.dat
> FILE : Saved security.dat
> RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1384, room 16
> tail 8
> chksum 0x2d
> csum 0x2d
> v09826c6d
> ~ld
>   
> 
> INIT : Booting version: mega-20180114
> INIT : Warm boot #6
> FS   : Mounting...
> FS   : Mount successful, used 75802 bytes of 957314
> INIT : Free RAM:25808
> INIT : I2C
> INIT : SPI not enabled
> WIFI : AP Mode disabled
> WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
> WIFI : Could not connect to AP!
> WIFI : AP Mode enabled
> WD   : Uptime 0 ConnectFailures 0 FreeMem 20576
> 

So I think PUYA P25Q80H is a bad choice of flash IC for esp8266

psy0rz, 2018-01-15:
ah thanks! since it seemed to work correctly with the legacy ESPEasy (that didnt use SPIFFS), it still might be worth finding out if there is a software workaround/bugfix possible as well.

thanks for shedding a light on it.

TD-er, 2018-01-15:
Should we log the flash info on boot? Just to simplify debugging on issues like this?

psy0rz, 2018-01-15:
is there a way to detect the flashissues? (besides random read errors)

TD-er, 2018-01-15:
The CRC pull req. may help I guess?

moo2089, 2018-01-15:
hi grigorygn,

good to read about memory compatibility issues, as it was also suspecious for me and ordered same flash chip you mentioned (WINBOND 25Q32FVSIG (4MByte)), in beginning of this month.
To be honest, choosing flash type was a gamble from my side:)
many thanks for letting us know!

psy0rz, 2018-01-15:
https://github.com/esp8266/Arduino/issues/4061#issuecomment-357810624

TD-er, 2018-01-15:
If my theory is correct, then it is just a matter of "programming" the flash chip with a signature to make it work again.
This could then be either patched in the ESP core library, or we could add a routine to check at first boot whether the flash size meets the magic string and if not correct it. Then after a reboot it might work again.

But then again it is just a theory, not tested yet.
I will try to create a sketch tomorrow, but it may destroy the flash if it is not the problem.

TD-er, 2018-01-16:
I created a little test (cannot test myself) by patching the ESP8266 core file esp.cpp:
In this test, getFlashChipSize() calls getFlashChipRealSize() to make sure all calls return the correct value, even when all bytes in the flash are set to 0.

```C++
uint32_t EspClass::getFlashChipSize(void)
{
    uint32_t data;
    uint8_t * bytes = (uint8_t *) &data;
    // read first 4 byte (magic byte + flash config)
	return getFlashChipRealSize();
    if(spi_flash_read(0x0000, &data, 4) == SPI_FLASH_RESULT_OK) {
        return magicFlashChipSize((bytes[3] & 0xf0) >> 4);
    }
    return 0;
}
```

Please test this version on the affected ESP modules.  (and I really hope the library files are all recompiled)
https://www.dropbox.com/s/wq9bgt17oan3s3n/Issue-650_ESPeasy_normal_ESP8266_1024_flashDetectTest.rar?dl=0

[31 more replies]

closed 2018-11-06 by Grovkillen