letscontrolit/ESPEasy #651: PCA9685 - Mode Setting

ShardanX, 2017-12-21:
The PCA9685 is an external PWM chip.
It can be set to different modes for direct LED use or for use with MOSFET or MOSFET/IGBT driver chip.
These modes are not accessible from the task setup in ESPEasy.

As LED stripes and such are daily used items to day this should be changed.

In the plugin init there is this line:
`Plugin_022_writeRegister(PCA9685_MODE2, (byte)0x10); // set to output`
It sets the output to invertzed and open drain configurtation.

I suggest to use two checkboxes for the mode2 settings INVRT (Bit 4) and OUTDRV (Bit 2).
In the init create a variable mode2set for examle.
If OUTDRV is checked do a mode2set=mode2set + 4  (Sets bit 2 to 1)
If INVRT is checked do a mode2set=mode2set + 16  (Sets bit 4 to 1)
Then write the mode2set to the MODE2 register of the PCA9685

This makes the use of FET's and driver chip possible. They need non inverted output, direct driving of a MOSFET needs totem pole config of the outputs, some drivers might need that too.

ShardanX, 2017-12-27:
Chnged Title for clarifying that the PCF8547 is included

psy0rz, 2017-12-31:
People watching this: Vote up the features or bugfixes you want like to see fixed first. use the thumbs UP arrow for that on the first comment. This way we can determine where to start, because we have LOTS of  requests. :)

TD-er, 2017-12-31:
I received my modules last week, so I might be looking into this.

Elektrofreak, 2019-08-22:
Any updates on the mode2 settings? My questions are:

- (How) is it possible to change the Mode2 register to totem pole?
- Which is the initial configuration during boot phase of the ESP and after initialization of the PCA?

Two more questions wich are not directly related to this thread and which could be moved into additional requests if useful:
- How is it possible to change the frequency via web command?
- Is it possible to set the output to 100% on ("real" on and not just 4095/4096)?
- Is it possible to implement a phase shift (e.g. shift each channel by 1/4, 1/8 or 1/16 period)?

Thank you very much for your effort!

tonhuisman, 2022-04-18:
The Mode2 value can currently be set from the Plugin configuration, so this has been 'quietly' added. This issue can be closed.

closed 2022-04-18 by TD-er