letscontrolit/ESPEasy #654: [memory usage] Device combobox takes enormous amount of memory

TD-er, 2017-12-23:
The device combobox with the listed devices takes about 4kB of memory.
For some device pages with lots of information, it has become too much.
For instance the OLED framed plugin - when active- can hardly load anymore and may lead to several crashes. Especially when the scrolling on the OLED display is enabled and more than 4 lines of text are available. (N.B. mega branch)

So perhaps we can generate the combobox with devices at compile time and store it in flash?
Or generate those things in real time when sending a HTTP chunk?
Or disable the plugin when settings are being altered?

psy0rz, 2017-12-28:
perhaps, or even get the list with a seperate ajax call?

TD-er, 2017-12-28:
I was thinking about generating part of the combo box only once and store it in flash.
I am still thinking about it and have not yet a clear answer to it.

psy0rz, 2018-01-02:
yeah ah you mean, generate it runtime and store it in spiffs?

TD-er, 2018-01-02:
Yes something like that.
Still some issues like detect changes, but that's the Idea.

psy0rz, 2018-01-02:
yep, but does that help? we still would have to load that file and that still takes 4kb of memory?

4k wouldn't be a problem if we create a extra GET handler for it that only creates and returns this list.

TD-er, 2018-01-02:
There is also the retrieving of the plugins, sorting them, converting to HTML (and adding the selected part), etc.
We could make a single function in the webserver class to stream it live without keeping it as a string in memory.
Just a single function that reads from flash, add wrapping HTML and get the selected item number as input and stream that directly to the web-client.

psy0rz, 2018-01-02:
perhaps..still its pretty complex and i wonder how much it eventually saves.

TD-er, 2018-01-02:
I will try and test myself.
You don't see any issues I may not see yet?

psy0rz, 2018-01-02:
ah the function would not cache it to flash? then i see no issue.

TD-er, 2018-01-02:
No, flash is for reading, not for writing (often) :)

psy0rz, 2018-01-02:
ok, i'm setting up a build "bot" to trigger nightly builds for the mega-branch. should make testing easier for users. (will make a forum post about this as well)

TD-er, 2018-01-03:
Could you please change the date order in the mega builds to yyyymmdd ?
Having a mega branch Autobuild is very helpful.

psy0rz, 2018-01-03:
Ok i will.

psy0rz, 2018-01-10:
i just though of a better solution: Just create a separate "Add device" page for empty tasks. 

After that we wont need to show the combobox on the Edit page. We only need to add a delete button to delete a device.

TD-er, 2018-01-10:
Sounds like a plan :)

psy0rz, 2018-01-10:
this was pretty easy to fix and we had a lot of memory issues, so moved to v2.0.

commit fe194492 (2018-01-10): changed devicecombobox handling to save a lot of memory on device page. fixes #654 #676 and could be triggered by #683 in some cases.
closed 2018-01-10 by psy0rz