letscontrolit/ESPEasy #655: output BMP280 sensor

JPBreestraat, 2017-12-23:
I installed a bmp280 next to a dht11. The output of the BMP280 sensor is a bit offset.
[image]

is there a way to correct it? ( for the dutch in jip en janneke taal)

thank you

TD-er, 2017-12-23:
Het kan ook in het Nederlands, indien noodzakelijk ;)
With a temperature (and pressure) offset this big, there is something else wrong.
Those BMP280 sensors may have a temperature offset of just a few degrees, but the pressure is actually spot on. 565 hPa is not spot on, unless you have it at quite some height above sea level. :)

But you are running Tasmota and this is the ESP-easy guthub :)
Tasmota is [next-door](https://github.com/arendst/Sonoff-Tasmota)
You can also try ESPeasy and see if the problem is with the hardware or the software.

ShardanX, 2017-12-23:
On top it might be better to place support questions in the forum, not on github.

JPBreestraat, 2017-12-23:
I was looking for an anwser on the Easyesp at the same time i was on Tasmota. wrote the question on the wrong forum :)

closed 2017-12-23 by JPBreestraat