letscontrolit/ESPEasy #656: 20000 - Mega32 Analog input issue (Plugin2)

redskinhu, 2017-12-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Setup the Analog input device on the devices tab on the ESP Easy webpage.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Permanent problem.

### Expected behavior
The 2nd calibrabration point max value is 4095 for ESP32.

### Actual behavior
The 2nd calibrabration point max value is 1023 (which is belongs to ESP8266).

### System configuration

Hardware:
ESP32

Software or git version:
Build: 20000 - Mega32

--

Grovkillen, 2017-12-23:
ESP32 support is in a very early stage.

redskinhu, 2017-12-23:
Hello

Thank you for the reply.

Can i download the source files?
I think i can modify it.

Thx
Tony

Grovkillen, 2017-12-23:
The source is the same as for ESP8266. Only compiled differently.

redskinhu, 2017-12-23:
Thanks

I see i need to dig in to the subject...

I like it, it is a very useful sw.

ShardanX, 2017-12-24:
Besides the limitation of the calibration point the ESP32 seems to have much more ADC channels then the ESP8266. 
It will need a general revamp of the plugin if we want to make use from the inputs.
This should include:
- detecting or setting the chip type (ESP32 | ESP8266)
- setting the limits on calibration
- choosing the ADC input channel on ESP32 only.

Merry Christmas to all of you.
Shardan

redskinhu, 2017-12-25:
A bunch of good ideas.
please....... :-)
for christmas

psy0rz, 2017-12-31:
Although ESP32 is pretty untested right now, we still should collect all the related issues. Till now it seems ESP32 support is easier than I initially suspected. (mainly because the awesome esp/arduino guys are fixing the libraries)

Perhaps support for more ADC inputs should be a seperate feature request.

People watching this: Vote up the features or bugfixes you want like to see fixed first. use the thumbs UP arrow for that on the first comment. This way we can determine where to start, because we have LOTS of  requests. :)

redskinhu, 2018-04-03:
Hi
Is it possible to fix this before compile new public ESP32 firmware?

I compiled and uploaded but it drops some kind of PID reated error and wont start.

Please.

Thanks T

TD-er, 2019-11-04:
This has been fixed a while ago, so I will close this issue.

closed 2019-11-04 by TD-er