letscontrolit/ESPEasy #659: Reature request.  Selectable rulesets.

Budman1758, 2017-12-24:
The ability to select a different set of rules would be a nice feature. My example is an irrigation controller that uses different rules for a summer vs winter schedule. In other words a simple way to turn on or off each rules page much like devices have an "enable" tick box.  This seems to be a popular idea on the forum.

Obviously one can change the rules by cut and paste but a simple way to "throw a switch" would be a nice addition. There are 4 "pages" of rules so I envision a tick box on each page that enables or disables them.

TD-er, 2017-12-24:
That would be a nice feature indeed, to enable/disable rules.
I think the enabled/disabled mode of rules can also be used to dynamically switch them.
For example enable/disabling them from other rules (time/date based) or based on communication from message brokers (e.g. MQTT)
Then you can enable a chain of rules by a single event, like "no-one at home, increase sensitivity" or something like that.

Grovkillen, 2017-12-24:
It's a great idea. More info on the forum: https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=3911

psy0rz, 2017-12-31:
sounds good indeed :)