letscontrolit/ESPEasy #663: Feature Auto AP disabled by jumper

druckgott, 2017-12-27:
Hallo, to habe not a problem if your wifi is disabled because you use the time to switch off for example a fritzbox, and then the esp goes to an ap and anybody else can connect to it and reconfigure it, it will be quite nice if it's possible to set a jumper with a cable between one pin gnd or connect TX and rx so that the ap mode is not possible and it's search only for the configuration ssid which is already saved on first instalation.

TD-er, 2017-12-27:
There is already a pull request, for the following version, which is more forgiving to unavailable wifi.
See #639 for the WiFi connection issues.
In the Mega branch there is also a pull request #653 to allow only access from a subset of IP-addresses.
Is that flexible enough?

druckgott, 2017-12-27:
OK. It looks good but I think to disable the accesspoint option over a hardware function is much more secure? It's more for security reason if you want to switch for example a garage door.

TD-er, 2017-12-27:
OK, so your suggestion is to disable web-access and access-point by a switch?

Sounds like a valid security feature :)

druckgott, 2017-12-29:
Jo exactly

ghtester, 2018-11-09:
It would be really nice to have the "AP mode disable" option available in Config. For me no jumper is necessary, just a checkbox for disabling AP mode would be great.