letscontrolit/ESPEasy #671: XY positioning in OLED module is not correct

linker3000, 2018-01-02:
### Steps to reproduce
* Compile latest code (as of this posting date) using PlatformIO
* Configure ESPEasy to connect to an MQTT broker (Actually used Mosquitto/Node-Red)
* Define a device as "Display - OLED SSD1306" and set Display size to 128x32
* Send MQTT message "oled,0,0,X" to /DEVICENAME/cmd

### Expected behavior
Message strings should be placed at the specified absolute value.

### Actual behavior
* X appears at end of the first row of the display
* Subsequent sends of the same message, put an X at the end of the next row and so on. Same result with longer strings - they are not placed at the specified absolute value.
* Specifying message payload as "oled,0,1,X" DOES put FIRST message in the right place, but subsequent messages are on next row etc.
* OLED clear command works as expected

### System configuration

Hardware: https://www.banggood.com/Wemos-TTGO-ESP8266-0_91-Inch-OLED-For-Arduino-Nodemcu-p-1205904.html?cur_warehouse=CN

Software or git version: Compiled from source on this day - Web interface shows a blank for 'git version'.

TD-er, 2018-01-02:
The 128x32 OLEDs have some strange memory layout.
Only the odd (or even?) lines are used.
This makes them somewhat compatible with the 128x64 displays.

linker3000, 2018-01-02:
Using 128x64 settings for these displays will only show alternate rasters of the font, so everything looks squished.

From a few quick additional checks, the origin (on these 128x32 displays at least) is 1,1 and not 0,0, and I can get consistent placing on that basis, so either the docs are wrong of the code needs a quick check over.

tonhuisman, 2022-04-18:
This issue is no longer valid, AFAICS, so can probably be closed.

closed 2022-04-18 by TD-er