letscontrolit/ESPEasy #672: Allow timeout for sending PWM signal to servos

TD-er, 2018-01-02:
As discussed in this thread on the forum: https://www.letscontrolit.com/forum/viewtopic.php?p=22045#p17516

Suggested implementation:
Allow 2 thresholds to get 3 ranges, with for each range a timeout to keep sending the last value.
Of course changes should be sent immediately.

Should the timeout be in msec or in samples?

psy0rz, 2018-01-02:
People watching this: Vote up the features or bugfixes you want like to see fixed first. Use the thumbs UP arrow for that on the first comment. This way we can determine where to start, because we have LOTS of requests. :)

Maus34, 2018-06-01:
Hi,
I would like to have msec. I will use this feature for a ventilation flap that operates about 2...10 times per day. For the rest of the time the friction of the servos is high enough to hold the flap in position.
Thank you.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

TD-er, 2023-09-04:
Not 100% sure the timeout of the PWM is present.
You can send a specific (documented) value to the servo command to stop sending the PWM command and thus keep the servos from actively correcting their position and thus heat up (burn out?)

Since this feature is present, I guess there is no need for a timeout option for the servo command as it is possible to 'quiet' the servos.

closed 2023-09-04 by TD-er