letscontrolit/ESPEasy #673: Strange switch behaviour when enabled "inversed logic" and "send to controller" at the same time

gelitoam, 2018-01-03:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

Hello.

I've noticed that when I connect a switch (tested and working ok, can see value changing 1/0 in Devices tab), then enable "Inversed logic" (again working ok, now I see 0/1) and then enable "Send to controller" (Domoticz/MQTT) the value locks to 1 and does not change despite the physical switch position (and Domoticz only receives "On" messages).

If then I disable "Inversed logic" everything works ok, value changes again 1/0 and Domoticz receives "On"/"Off". If I keep "Inversed Logic" but disable "Send to controller" the switch works as expected (value changes 0/1 and of course nothing is sent to Domoticz).

Is this a bug or am I misunderstanding something?

Best regards.

### Steps to reproduce
Switch input with "Inversed Logic" and "Send to controller" enabled.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes.
### Expected behavior
When switching, value should change 0/1 and Domoticz should receive "Off"/"On"

### Actual behavior
Value keeps 1 after switched for the first time, and Domoticz receives "On" every subsequent switch action.

### System configuration

Hardware: ESP12F

Software or git version:

TD-er, 2018-01-04:
I've done a quick look at the code and here is the bug:
https://github.com/letscontrolit/ESPEasy/blob/e4d68b9d0e7a477cd57ae3d6eabc2265b87b5964/src/_P001_Switch.ino#L117-L146

It looks to me like the check for Settings.TaskDevicePin1Inversed[event->TaskIndex] should be done always when getting the current state.
So I guess it should be something like a single function to get the current state, which will always flip based on the invert setting.
But I have to look into it a bit more to make sure nothing else will break ;)
At least it makes sense no update will be sent to the controller.

TD-er, 2018-01-04:
OK, the main problem was a bit surprising.
The switch type (switch or dimmer) was stored incorrect using default settings.
You actually had to change the type (and save) to get it to work.
Problem was the default value stored in flash is a "0", but the only valid options were "1" and "2".

I now changed it to "0" and "1".
Anyone who was using a switch and does upgrade, now has to make sure to check the switch type and save again.
New added switches will now work using the displayed settings.

Also I rewrote the code to make it better readable, since it was quite hard to understand.

closed 2018-01-05 by psy0rz

jannnfe, 2018-02-22:
In Version v2.0-20180217 and higher I still have the problem. It took me a long time to reproduce it, but I'm sure the combination of Domoticz controller and Inversed logic + send to controller is not working properly. The error as described above persists.