letscontrolit/ESPEasy #674: Esp easy update temperature to domoticz generate a reboot

Tixav, 2018-01-04:
I recently update a nodemcu from R147 (which is working for 1 year) to v2.0.0-dev12 then to v2.0.0-dev13 but the result is the same : 
I setup a controller (domoticz)
I setup a ds18B20 device (set 12 bits resolution, idx set, ... )   
[https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3959](url) 
no issue it is working well, but if I checked **Send to Controller**, then the nodemcu is rebooting randomly , I also check with a weemo D1 mini it is the same !

for the moment, I use this counter measure (by rules), and it is working well without random reboot:
`
//when temp is updated from device setup page
on temp#Temperature do
 SendToHTTP 192.168.1.4,8080,/json.htm?param=udevice&type=command&idx=510&nvalue=0&svalue=[temp#Temperature]
endon
`

TD-er, 2018-01-04:
What controller are you using? e.g. http/MQTT?
How is the controller set up?
What values do the 1wire device give? (e.g. NaN)

Also screenshots of the device tab and/or controller would be nice.

Tixav, 2018-01-04:
What controller are you using? e.g. http/MQTT?: I setup Domoticz Http controller
How is the controller set up? see the printscreen (I update the topic) @ [https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3959](url)
What values do the 1wire device give? (e.g. NaN) : Correct values until reboot
Also screenshots of the device tab and/or controller would be nice. : [https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3959](url)

TD-er, 2018-01-04:
It also happens on the switch input.
When the Domoticz controller is active and you press a number of times the button, the ESP device will reboot.
Looks a bit like it is trying to send a number of messages, while the last one is not yet finished.
My test here was with Domoticz MQTT as active controller.

TD-er, 2018-01-05:
Just before the crash, it reports Exception (2).
```
Exception (2):
epc1=0x3fff3a48 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff3a48 depc=0x00000000

ctx: cont
sp: 3fff3bf0 end: 3fff43c0 offset: 01a0
```
Taken from [this list](https://links2004.github.io/Arduino/dc/deb/md_esp8266_doc_exception_causes.html), which appears to be: 

> InstructionFetchErrorCause | Processor internal physical address or data error during instruction fetch

psy0rz, 2018-01-05:
sounds serious, ill try to reproduce it as well.

TD-er, 2018-01-05:
I was just playing around with running background tasks just to make sure the watchdog is not kicking in. with multiple controller messages in a short time.

So I added something like this:
```C++
void sendData(struct EventStruct *event)
{
  if (timePassedSince(lastSend) < 1000) {
    // Frequent subsequent calls to the controller.
    // Perform yield first to make sure the wdt reset will occur.
    //yield();
    delay(1);
  }
```

Somewhat inspired by #676 and also what was discussed here: https://github.com/esp8266/Arduino/issues/1997#issuecomment-261127133

But adding that code will lead to much more frequent crashes.
The exception(2) (InstructionFetchError | Attempt to execute instructions from an illegal address, or read error when fetching instruction) is some kind of indication a function call is being made to something that is no longer (or not yet) there.
Something like a pointer dereference to something that's being destroyed or not deleted properly.
For example some inherited class whose destructor is not virtual.
It could also be the result of some buffer overflow, or a string pointer without a 0 termination.

So lots of possibilities. Any suggestion on how to analyse the stack?

psy0rz, 2018-01-06:
I think there was some stack analyser for platformio or arduino GUI, but i forgot how it was named. It could translate the stack dump to a nice backtrace.

TD-er, 2018-01-06:
Hmm, when built using the Arduino IDE, I cannot get it to crash.
But using the same source, built using PlatformIO and uploaded via Atom, I get a crash after just a few presses on the button (using the flash button on the NodeMCU)

It uses the same config, since that was not changed between flashes.

Problem is, I now have a stacktrace via Arduino IDE, but it doesn't make sense, so I guess the images are actually different in build settings. (the .elf file is significantly smaller when built using PlatformIO)
I don't see any line numbers in the stacktrace.
I use the ESP Exception Decoder, described here: http://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

```
Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
Decoding 38 results
0x4021067a: PubSubClient::loop() at ?? line ?
0x40210512: PubSubClient::loop() at ?? line ?
0x40211fc5: ESP8266WebServer::handleClient() at ?? line ?
0x40236b89: backgroundtasks() at ?? line ?
0x40236bc0: delayBackground(unsigned long) at ?? line ?
0x40238829: sendData(EventStruct*) at ?? line ?
0x4023915c: Plugin_001(unsigned char, EventStruct*, String&) at ?? line ?
0x40101b41: trc_NeedRTS at ?? line ?
0x40101cd4: trc_NeedRTS at ?? line ?
0x4010020c: _umm_free at umm_malloc.c line ?
0x402023f0: String::changeBuffer(unsigned int) at ?? line ?
0x4010020c: _umm_free at umm_malloc.c line ?
0x4010020c: _umm_free at umm_malloc.c line ?
0x40100688: free at ?? line ?
0x40202390: String::~String() at ?? line ?
0x40229a14: callback(char*, unsigned char*, unsigned int) at ?? line ?
0x401048f9: ets_timer_disarm at ?? line ?
0x40106f01: __udivsi3 at d:\ivan\projects\arduinoesp\toolchain\dl\gcc-xtensa\build-2\xtensa-lx106-elf\libgcc/../../../libgcc/config/xtensa/lib1funcs.S line 540
0x40107114: millis at ?? line ?
0x4010053d: realloc at ?? line ?
0x40216171: timePassedSince(unsigned long) at ?? line ?
0x402023f0: String::changeBuffer(unsigned int) at ?? line ?
0x4020243f: String::reserve(unsigned int) at ?? line ?
0x40202475: String::copy(char const*, unsigned int) at ?? line ?
0x402024ca: String::String(char const*) at ?? line ?
0x40202390: String::~String() at ?? line ?
0x4021d072: PluginCall(unsigned char, EventStruct*, String&) at ?? line ?
0x40203430: esp_yield at ?? line ?
0x402034a8: yield at ?? line ?
0x40210580: PubSubClient::loop() at ?? line ?
0x40203430: esp_yield at ?? line ?
0x402384a1: run10TimesPerSecond() at ?? line ?
0x4023de91: loop at ?? line ?
0x4020347c: loop_wrapper() at core_esp8266_main.cpp line ?
0x40205e14: cont_norm at cont.o line ?
```

TD-er, 2018-01-06:
I added the option "-g3" to the PlatformIO build flags and now I believe I have some stacktrace:

```
Exception 2: InstructionFetchError: Processor internal physical address or data error during instruction fetch
Decoding 45 results
0x4020e3ba: PubSubClient::loop() at C:\Arduino\ESPeasy\arduino-1.8.3\portable\sketchbook\ESPEasy_Gijs\ESPEasy/lib\pubsubclient\src/PubSubClient.cpp line 590
0x4020e256: PubSubClient::loop() at C:\Arduino\ESPeasy\arduino-1.8.3\portable\sketchbook\ESPEasy_Gijs\ESPEasy/lib\pubsubclient\src/PubSubClient.cpp line 590
0x40211f79: ESP8266WebServer::handleClient() at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 299
0x40236a19: backgroundtasks() at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40236a50: delayBackground(unsigned long) at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40238680: sendData(EventStruct*) at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40238fa4: Plugin_001(unsigned char, EventStruct*, String&) at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x401040fc: lmacTxFrame at ?? line ?
0x40104062: lmacTxFrame at ?? line ?
0x40100793: ppProcessTxQ at ?? line ?
0x40100e32: pp_post at ?? line ?
0x40103c67: lmacMSDUAged at ?? line ?
0x401035c6: lmacRecycleMPDU at ?? line ?
0x4010020c: _umm_free at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x402023f4: String::changeBuffer(unsigned int) at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x4010020c: _umm_free at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010020c: _umm_free at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40202394: String::~String() at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x40229908: callback(char*, unsigned char*, unsigned int) at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40203434: esp_yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x402034ac: __yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x402034ec: optimistic_yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x4020bd27: WiFiUDP::parsePacket() at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiUdp.cpp line 283
0x40236642: checkUDP() at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x401048fd: ets_timer_disarm at ?? line ?
0x401048fd: ets_timer_disarm at ?? line ?
0x4020e26d: PubSubClient::loop() at C:\Arduino\ESPeasy\arduino-1.8.3\portable\sketchbook\ESPEasy_Gijs\ESPEasy/lib\pubsubclient\src/PubSubClient.cpp line 590
0x4010053d: _umm_realloc at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x402023f4: String::changeBuffer(unsigned int) at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x40202443: String::reserve(unsigned int) at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x40202479: String::copy(char const*, unsigned int) at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x402024ce: String::String(char const*) at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x40202394: String::~String() at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.cpp line 688
0x4021cffa: PluginCall(unsigned char, EventStruct*, String&) at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40203434: esp_yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x402034ac: __yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x4020e2c0: PubSubClient::loop() at C:\Arduino\ESPeasy\arduino-1.8.3\portable\sketchbook\ESPEasy_Gijs\ESPEasy/lib\pubsubclient\src/PubSubClient.cpp line 590
0x40203434: esp_yield at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x402382f5: run10TimesPerSecond() at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x4023dcbd: loop at C:/Arduino/ESPeasy/arduino-1.8.3/portable/sketchbook/ESPEasy_Gijs/ESPEasy/src/_P047_i2c-soil-moisture-sensor.ino line 515
0x40203480: loop_wrapper at C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 56
0x40205df0: cont_norm at cont.o line ?

```

TD-er, 2018-01-07:
Some update. Also related to what @s0170071 already noticed in #676 

Things to rule out: (at least not related to this issue)
In PubSubClient.cpp I added 2 checks to make sure no out-of-bound issues with the buffer were causing these crashes:
Changed:
(line 247) `  } while ((digit & 128) != 0); `
To:
`    } while ((digit & 128) != 0 && len < (MQTT_MAX_PACKET_SIZE -2)); `

And similar:
Changed: 
(line 528)    `while (*idp) { `
to:
`    while (*idp && pos < (MQTT_MAX_PACKET_SIZE - 2)) { `

That made no difference.

In SendData() (Controller.ino) I removed
`delayBackground(delayms);`

Then you only get the controller to crash when pressing the switch intensively for a number of seconds.
This results in a panic, with the watchdog kicking in:
`Panic C:\users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_main.cpp:98 __yield
`

So I added a `yield();` , but that made no difference.

The code is also rather fishy about checking only controller 0 for being enabled or having MQTT use.

I think I can fix the issue with the switch by delaying consecutive updates and only sending the last one when pressing the switch very frequently. But I think the real problem is with the sendData function and the background services. So other plugins may experience similar issues.

s0170071, 2018-01-07:
Have you tried core 2.3.0 ? 

https://github.com/letscontrolit/ESPEasy/issues/674#issuecomment-355787811
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

TD-er, 2018-01-07:
I'm only running on core 2.3, since that's the version with ESPeasy

s0170071, 2018-01-07:
My crash was from an Arduino IDE build. So the problem is still there. Had sth. similar a while ago. In the end it was a uninitialized variable in an interrupt function. One compiler setting initialized it to zero, the other didn't.

TD-er, 2018-01-07:
Hmm, that may be a good pointer to start looking, since in the Arduino IDE the crash was at least much harder -if not impossible- to reproduce.

psy0rz, 2018-01-10:
If you set a message delay of 30000ms at the advanced settings page, this bug gets triggered every time.
(i'm using switch plugin with domoticz mqtt)

I'll try to figure out whats going on and i'll test your PR as well. Also i wonder what happens if a MQTT message arrive and the mqtt callback is called when doing a delaybackground somewhere.

psy0rz, 2018-01-10:
i added a bunch of debug statements and indeed its crashing on the mqtt stuff in background tasks.

the easy way out would be to move MQTTclient.loop() to loop() so that its never called from delayBackground()

s0170071, 2018-01-10:
Don't need mqtt for a crash. See #676. I suspect the stack. Does anyone of you know how where the sp is located?
@td-er do you think initializing / zeroing the whole memory (ram) is done anywhere in the code already?

TD-er, 2018-01-10:
@psy0rz Good find. I remember looking at that part and finding it a bit strange, but didn't trigger a "oh-no" reaction ;)

I also noticed on the stacktrace that the loop was being called from loop(), but I couldn't see why. So I was always a bit unsure about the added value of the stacktrace.

@s0170071 I don't think it is cleared, since some things survive a reboot.
I don't think it is necessary to clear it, since the uninitialized part should never be read anyway ;)
But I think it is a good addition to allow to clear the configuration flash (all except the firmware) to allow a clean start and make sure no left over configuration is causing issues.

psy0rz, 2018-01-10:
@s0170071 its still possible that they have enabled a mqtt controller in #676 . (especially since it seems to be fixed after removing the delaybackground there)

i'll see if i have that display to make sure.

s0170071, 2018-01-10:
"They" was me. And I did not enable anything other than that display. At least not that I am aware of....
Remember: press EDIT button of that display in the web interface the very moment scrolling happens. It's good practice to push the mouse button before and just release it ... You'll figure it out :-)

psy0rz, 2018-01-10:
ah ok sorry. was hoping it was the same underlying issue. :)

psy0rz, 2018-01-10:
ok we fixed some issues that might be related. can you try again? the next nightly v2.0 release should have the fixes. if its still not fixed let me know. i'll also try to reproduce it with http and the ds18 here. (will add it to the automated test suite)

psy0rz, 2018-01-12:
should be fixed now, please check tonights nightly's

closed 2018-01-12 by psy0rz