letscontrolit/ESPEasy #676: Framed Display crashes ESP

s0170071, 2018-01-05:
### Steps to reproduce
compile master with core 2.4.0 (fix the min max issue as mentioned in the forum)
set up a framed display, activate it. 
interact with the web interface the very moment scrolling takes place.

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
y

### Expected behavior
Tell us what should happen?
 
### System configuration
Wemos D1 mini.

ps.
in the function 
`void display_scroll(String outString[], String inString[], int nlines, int scrollspeed)`
there is a call to 
`backgroundtasks();`
disable it and the issue is gone. 
Question is: what are the drawbacks ?

s0170071, 2018-01-05:
forgot to mention: The crash does not happen on any web interaction. Go to devices and click "edit" of the framed display.

Serial output shows then:
`

INIT : Booting version: (custom)
INIT : Warm boot #13
FS   : Mounting...
FS   : Mount successful, used 76053 bytes of 957314
INIT : Free RAM:24872
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_11 with address 192.168.4.1
WIFI : Connecting MNET attempt #1
WIFI : Connected! IP: 192.168.1.32 (ESP_Easy_11)
WD   : Uptime 0 ConnectFailures 0 FreeMem 20744
WD   : Uptime 1 ConnectFailures 0 FreeMem 20032

Panic /home/john/ArduinoPortable/arduino-1.8.5_ESP2.4.0/portable/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/core_esp8266_main.cpp:133 loop_task

ctx: cont 
sp: 3ffffdd0 end: 40277f09 offset: 01b0

>>>stack>>>
3fffff80:  4025a812 3ffee9f0 00000000 3fff2b00  
3fffff90:  00000000 3fffdad0 3fff2af8 4023d371  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffffb0:  40000e19 40001878 00000002 00000000  
3fffffc0:  3fffff10 aa55aa55 0000001d 401043c8  
3fffffd0:  401043ce 00000002 00000000 00000580  
3fffffe0:  4010000d 00000568 00000000 40001100  
3ffffff0:  4000044c 00000000 00000000 0b675f24  
40000000:  00000000 00000000 00000000 00000000  
40000010:  46007200 0000fffe 00000000 00000000  
40000020:  00003310 00000000 00000000 00000000  
40000030:  46004100 0000fffe 00000000 00000000  
40000040:  00000000 00000000 00000000 00000000  
40000050:  29ffd112 31613951 e820f695 a0323003  
40000060:  71490338 000003a0 00000000 00000000  
40000070:  46004140 0000fffe 00000000 00000000  
40000080:  00000806 00000000 40000000 2222211f  
40000090:  e0000000 400000f3 4000e328 00000000  
400000a0:  00000000 e400000c 13050013 21002010  
400000b0:  eb30fff5 74303003 2356428c 00620200  
400000c0:  20fff221 612013e7 fff02100 61fff151  
400000d0:  030cfff1 6650027d 00058610 00000000  
400000e0:  00506340 f03d0020 3350f03d 14b3b6c0  
400000f0:  70417470 13673440 506340e6 f6c03350  
40000100:  2000edb3 ffe25100 7220030c 34407020  
40000110:  5050e340 7470c033 f0b3f641 21002030  
40000120:  02acffde 12480238 c2222258 0fb3470c  
40000130:  554b0568 334b0369 46f43347 0000fff9  
40000140:  56fe0356 2000fdd5 002e0500 00000000  
40000150:  00000000 00000000 00000000 00000000  
40000160:  00000000 00000000 00000000 00000000  
40000170:  00000000 00000000 00000000 00000000  
40000180:  00000000 00000000 00000000 00000000  
40000190:  00000000 00000000 00000000 00000000  
400001a0:  00000000 00000000 00000000 00000000  
400001b0:  00000000 00000000 00000000 00000000  
400001c0:  00000000 00000000 00000000 00000000  
400001d0:  00000000 00000000 00000000 00000000  
400001e0:  00000000 00000000 00000000 00000000  
400001f0:  00000000 00000000 00000000 00000000  
40000200:  00000000 00000000 00000000 00000000  
40000210:  00000000 00000000 00000000 00000000  
40000220:  00000000 00000000 00000000 00000000  
40000230:  00000000 00000000 00000000 00000000  
40000240:  00000000 00000000 00000000 00000000  
[... 712 lines cut ...]
40002ed0:  11c90261 02cd0109 010ae5c5 0008fffb  
40002ee0:  c0c002dd 0ae50582 c7c032d0 e48508b3  
40002ef0:  c042d00a c8f634c7 0821d811 10c11201  
40002f00:  0000f00d 29fff131 00f00d03 28ffef21  
40002f10:  00f00d02 260e3226 c2420f62 0cd30cf4  
40002f20:  83234002 b20cf00d c20cf00d 0000f00d  
40002f30:  3fffc708 28ffff21 00f00d02 09f0c112  
40002f40:  ffff0501 0173a042 1226f885 08226612  
40002f50:  220020c0 1226c420 1201081a f00d10c1  
40002f60:  420020c0 20c0c560 c4203200 c1120108  
40002f70:  00f00d10 420020c0 20c0c660 c5604200  
40002f80:  c1120108 00f00d10 09f0c112 fffac501  
40002f90:  c1120108 00f00d10 40002f88 3fffdde0  
40002fa0:  d9f0c112 e9010921 ed11c931 ffe03103  
40002fb0:  230202cd 20d44000 22082066 fd8500a1  
40002fc0:  ffdb31fd 13e923d9 6401e67c 0020c0f8  
40002fd0:  60c02052 20c01055 c0605200 a4164c0b  
40002fe0:  252c260b 26224c26 180c753c 720020c0  
40002ff0:  7780c020 0020c020 31e821d8 607203c9  
40003000:  0811c8c0 10c11201 a092f00d 0020c
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

`

TD-er, 2018-01-05:
This call to backgroundtask is comparable to yield(), isn't it?
This yield() should perform more time limited things like WiFi stuff, periodical stuff (multiple times per second tasks) and such.

s0170071, 2018-01-05:
So omitting it is not a solution but merely a hint on what is going wrong.

https://github.com/letscontrolit/ESPEasy/issues/676#issuecomment-355679259
Diese Nachricht wurde von meinem Android-Gerät mit K-9 Mail gesendet.

TD-er, 2018-01-05:
It is in ESPEasy.ino (line 1322).
It is indeed a very descriptive function name, which does what it says ;)

s0170071, 2018-01-07:
if you replace the backgroundtasks() with just webserver client handling it does also crash if you press the edit button while the display scrolls. 

```
   display->display();

    WebServer.handleClient();
    //backgroundtasks();
  }
```

I would love to try lwip2 but can't get it to compile yet, ESPEasy uses some functions which are not available with lwip2 any more.

TD-er, 2018-01-07:
In the mega branch, I updated the library for the OLED. So maybe you can also test with that one?

s0170071, 2018-01-07:
no success. 
Since I am fairly new to GitKraken and just to be sure: it really is the mega most current mega branch, right ?

[image]

TD-er, 2018-01-07:
Yep it is.
You can also run the pre-built versions from the Mega branch: https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20180104

s0170071, 2018-01-07:
I just added a printf at the end of the plugin
```
       break;
      }

  }
      Serial.println("ENDplugin");

  return success;
}

```

and two more in the scroll routine: 
```
    display->display();

    //delay(2);
   WebServer.handleClient();
   Serial.println(i);
  //  backgroundtasks();
    
  }
  Serial.println("finished scroll");

}

```

The Output without crash is: 
```

ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
A
0
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
finished scroll
ENDplugin
ENDplugin
ENDplugin
ENDplugin

```

and with crash: 
```
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
A
0
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin
ENDplugin

Exception (0):
epc1=0xc0277f1c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffddc end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff7c:  3ffeb168 4025a822 3ffeea10 00000002  
3fffff8c:  3fff2b20 00000000 3fffdad0 3fff2b18  
3fffff9c:  4023d36f 40000f49 3fffdab0 3fffdab0  
3fffffac:  40000f49 40000e19 40001878 00000002  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

;l
```

So: it does not finish the scroll loop but directly returns from the plugin call. Sounds an awful lot like a full stack to me. I googled a bit but could not get a hold of the SP register address. Might be worth monitoring.

TD-er, 2018-01-07:
I am currently writing default constructors for all the structs in ESPeasy.ino and I will create a commit for them. (you don't want to do that again, which is a painful cut'n-paste job.
When that's committed, you may want to have a look at it again.

TD-er, 2018-01-07:
@s0170071 I created pull request #682.

It is either fixed now, or has now become really hard to reproduce ;)

s0170071, 2018-01-07:
I cannot checkout the pull request. GitKraken shows 7 pending requests but when I click on it it would create a new one. When I do the same thing with the core git I can pick a pr and it checks it out.

TD-er, 2018-01-07:
As a test, you can try to copy the files from my changes to your local branch.
You have to realize they are yet based on the v2.0 branch.
https://raw.githubusercontent.com/TD-er/ESPEasy/be25aa552224e85fe8f432f3edbd5795be31009a/src/ESPEasy.ino

s0170071, 2018-01-08:
There is a function missing...

/home/john/Arduino/scetchbooks/ESPEasy_git/ESPEasy/ESPEasy.ino: In function 'void loop()':
ESPEasy:1004: error: 'timeOutReached' was not declared in this scope

TD-er, 2018-01-08:
That was in #621 
I think it is then the easiest to create a clone of my branch: https://github.com/TD-er/ESPEasy/tree/bugfix/v2.0_crash_switch

Maybe we should add some workflow/Git introduction to the wiki, since I also struggled a bit in the beginning.
In short, you should create a clone of the Git repository to your own account and on that one, create a branch, fix stuff, push back to your own account, create a pull request.
Only problem is to keep your own clone in sync with the original repository. That's also still the thing I'm struggling with and I haven't found a way of doing that in GitKraken.

s0170071, 2018-01-08:
oh man, sigh.
I tried to check it out as part of the ESPEasy repo but that doesn't work.
I managed to clone it into a different folder but now I need to change my folder shortcuts back and forth because the Arduino IDE forces .ino files to be in an folder thats called ESPEasy and not src. 
And the ESPEasy folder is linked to the clone of the mega branch.
Honestly, I spend about 90% of my time juggling files, installing and re-installing IDEs and so on.
I miss the times when you had a single source folder, decent data sheet and a JTAG debugger.
:-(

TD-er, 2018-01-08:
I have created a separate path for my Arduino builds. This contains a symbolic link to the src directory just so Arduino IDE sees an ESPEasy directory with .ino files and the other path (the git repository) is being used for GitKraken and Atom + PlatformIO.
In Linux it is really easy to create symbolic links (`ln -s`) and in Windows it is also possible. For example, see: http://forum.arduino.cc/index.php?topic=335588.0

But I think we should definitely create an issue and/or a forum topic to gather information on how to use a good workflow and should document this on the wiki.

s0170071, 2018-01-08:
Agreed. I would even go a step further and create a setup tool / script to have a "one click ready to use as described" environment. 
I do have four portable Arduino IDE's, one for each core version. Each with a different board URL. Not to forget to pick the right board, port, lwip,...
Gitkraken with two repos checked out. Then the shortcuts for the libs and the src folder for the ide: that's a real pain :-| . Usb driver.
Maybe platformio, an account for it. Settings for platformio.

Having such a setup tool would make it a pleasure to try a fresh setup if things become strange...
And it's a good starting point for additional operations and a documentation.

Mega bonus: debug. There are rumors that there is some sort of soft JTAG available. But setting this up would probably take its time. Or the gdb stub with visual gdb. But wait, wasn't this the purpose of platformio?
Ah, and there is also visualmicro with their soft debugger. haven't tried it yet.

You see, lots of work. And once you're done you get doubts about the directory structure.

https://github.com/letscontrolit/ESPEasy/issues/676#issuecomment-355938801

TD-er, 2018-01-08:
I found out the "hard" way last weekend that PlatformIO does not yet support ESP12e boards for debugging.
When you create an account for PlatformIO, the trial period is activated immediately. So do not create an account if you don't need the extra's, to save your trial period.
Since my account already exists for about 6 months, the trial had passed unused, so I took a 1 month subscription (should cost $11,99) and in the end it appeared to be about 45% more expensive (euro 11,99 + VAT) and after activation, it appeared the platform was not supported :(

psy0rz, 2018-01-09:
isn't this related to memory issues @mvdbro mentioned with core 2.4.0? It uses much more memory, which could explain those crashes.

psy0rz, 2018-01-10:
are you using mqtt?

psy0rz, 2018-01-10:
@TD-er thats too bad :( usually i work around stacktraces or use the arduino ide + exception decoder. (used it for 1 or 2 hard bugs in the past)

psy0rz, 2018-01-10:
added this to v2.0 since its probably an underlying core issue.

psy0rz, 2018-01-10:
i could reproduce this problem and was memory related. latest commit fixes it.

s0170071, 2018-01-10:
Just tried 2.0 with core 2.3.0 and 2.4.0rc2
good news first. There is no crash. 
bad news: 
1.the chunking issue is back (2.4.0rc2) - maybe was never gone in v2.0?, 
2. when I click device->edit it opens the edit page, but labels it "task settings" with no way to change the device (which is  "Generic - MQTT Import").
3. The display display 8 lines although only 3 are configured.

Didn't test any further. 
Just for my understanding... "2.0" is stable and "mega" is 2.0+improvements ?

psy0rz, 2018-01-10:
we've changed the way the Device type is selected. Once its selected, you need to click "delete" (bottom of the screen) to reset it so you can select another device. (this actually saves lots of memory)

Indeed 2.0 is stable, mega is 2.0 + new features. altough we're lagging a bit, we still need to merge those changes back to v2.0. (doing that as we speak)

so perhaps you could try with the megabranch after that to see if the chunking is ok again.

psy0rz, 2018-01-12:
all fixed now in both branches. if i only fill the first few fields, it works like expected.

closed 2018-01-12 by psy0rz

s0170071, 2018-01-12:
mega is working, 2.0 not. But the crash is gone.

TD-er, 2018-01-12:
The difference is that Mega is using a newer library version for the Framed OLED. (and a few small fixes)

psy0rz, 2018-01-12:
strange v2.0 is working with my oled display. i'm using the wemos oled display.

TD-er, 2018-01-13:
@psy0rz :  @s0170071 is compiling with 2.4 version of the ESP core

psy0rz, 2018-01-13:
ah ok..will add a ifdef to v2.0 so it gives a compiler errror