letscontrolit/ESPEasy #677: Sonoff Basic Error while reading/writing config.dat

jvanderkroon, 2018-01-06:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Flash espeasy -  branche mega commit d2e2522c0718effc809636ed914330fbe7b57236 - version normal_ESP8266_1024
connect to the wifi ap 
give the wifi credentials and save

Then i get the error:

Welcome to ESP Easy Mega AP
FS : Error while reading/writing config.dat in 966
ESP is connected and using IP Address: 
192.168.1.33

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
can read/write config.dat

### Actual behavior
Can't reading/writing config.dat

### System configuration

Hardware:
sonoff basic with ESP8266EX

Software or git version:
Build with Atom and platformio 2.8.0 , git 2.14.3

TD-er, 2018-01-06:
Are you really sure this Sonoff uses the ESP8266 and not the 8285?
Also please show the chip ID of the ESP and flash. (Tools -> Info)

Grovkillen, 2018-01-06:
+1 on that @TD-er

Grovkillen, 2018-02-10:
Did this get fixed with some commit?

psy0rz, 2018-02-26:
if it still doesn't work try the special PUYA firmware. (its in the same ESPEasy release zip file)