letscontrolit/ESPEasy #679: Random reboots because of controller?

Grovkillen, 2018-01-07:
My and some friends has done tests with no controller setup and units that randomly reboots 1-10 times per day with controller on number 1 is running with no problem with no controller active.

Can this be of use for the  developers?

psy0rz, 2018-01-09:
yeah probably if you have more info? :)

psy0rz, 2018-01-10:
what kind of controller was it? mqtt?

Grovkillen, 2018-01-10:
Yes OpenHAB controller (but used as generic MQTT since we don't have such a
controller, only removed first slash from publish/subscribe topics).

psy0rz, 2018-01-10:
i confirmed this bug

Grovkillen, 2018-01-10:
Ah great!

psy0rz, 2018-01-10:
ok can you try again, it's probably fixed now. if not let me know. :) tonight there will be a v2.0 nightly build (the first one :)

closed 2018-01-10 by psy0rz

Grovkillen, 2018-01-10:
Sweet! Will compile one right away.

Grovkillen, 2018-01-10:
And it's rolling...

Grovkillen, 2018-01-11:
And rolling, 11 hours no reboot. Seems like a solid fix, thanks @psy0rz