letscontrolit/ESPEasy #680: CAn you add plugin P170 (_P170_HLW8012.ino) for Sonoff Pow

diving91, 2018-01-07:
Hi,

Would it be possible to add the playground plugin to support Sonoff Pow in the ESP_Easy_mega-2018xxxx_dev_ESP8266_xxx builds ?

I've difficulties to build from the source code from Arduino IDE and this would be of great help. I think the Sonoff Pow is enough popular to add this plugin.

Thansk in advance
diving

Grovkillen, 2018-01-07:
Please lift this with the dev of the plugin, it's up to the plugin developer to make a PR.

diving91, 2018-01-07:
Thanks - I don't know how to do what you ask me ... soryy.
on the plugin code I see
`#ifdef PLUGIN_BUILD_TESTING`
which if I understand correctly mean that the plugin should be complied in the 2018xxxx_test_ESP8266_xxx builds ?
... this is not the case.

TD-er, 2018-01-07:
There are 2 repositories, The [url=https://github.com/letscontrolit/ESPEasyPluginPlayground]ESPEasyPluginPlayground[/url] and the ESPEasy repository.
The builds in the Mega branch are made from the ESPeasy repository and the P170 plugin is in the Playground repository.
That's why they are not being used in the automatic builds.

But @papperone made it and it seems there has been no development since september, so it might well be the plugin is stable enough to transfer it to this repository.

diving91, 2018-01-07:
From various forums, it seems the plugin is running OK.
so i might be about moving it from the playground repository to the 'main' repository.
That would really be great to avoid to recompile in order to have this plugin

psy0rz, 2018-01-10:
@papperone can you tell us if you know any known issues with this plugin? otherwise can you move it to the mega-branch and make a PR? 

Instructions on how to do this are here: https://www.letscontrolit.com/wiki/index.php/ESPEasyDevelopmentGuidelines (Adding plugins to the master repostory)

papperone, 2018-01-10:
hi, honestly I wanted to review this plugin to make some improvements but if it's considered stable I'll move it as it is to the mega branch and I'll make the improvement over there :)

psy0rz, 2018-01-10:
yes, just move it the mega branch and add it to DEVELOPMENT for now. unless you think its stable enough to move to TESTING.

papperone, 2018-01-10:
I committed it as P075 in teh "mega" branch.
it's tagged as "TESTING", plugin is running ok in my 2 units, the improvements I'll do are to enhance it and not to bugfix

papperone, 2018-01-10:
TRAVIS keeps giving errors which I can't understand how to fix it...

psy0rz, 2018-01-10:
#689

Grovkillen, 2018-10-23:
https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P076_HLW8012.ino

closed 2018-10-23 by Grovkillen