letscontrolit/ESPEasy #683: delayBackground(time) is not working properly and might cause crashes

TD-er, 2018-01-07:
See pull request #682 and issues  #679, #676 and #674 and [this topic on the forum.](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=3959)

Call to delayBackground() is now removed from the Controller::sendData(), which was probably the most frequently used call to this function.

This function is a wrapper to run backgroundtasks() often while waiting.

It is also called from:

- deepSleep (in Misc.ino)
- Command "Delay" for rules (in Command.ino)
- Controller "Generic HTTP" when sending data from a sensor with multiple values.
- Controller "Generic UDP"  when sending data from a sensor with multiple values.
- Plugin 015 TSL2561 I2C Lux Sensor when reading data (looks like "delay(...)" could work also here)
- Plugin 047 Moisture & Temperature & Light I2C Soil moisture sensor when reading data or changing settings (looks like "delay(...)" could work also here)

backgroundtasts() is called from:

- ExecuteCommand() for the debugging command "background"
- At the end of the main loop in ESPEasy.ino
- During scroll animation in Plugin 036: OLED SSD1306 display  (see issue #676 )

It looks like calling the backgroundtasks() too often may trigger something strange.

Also strange is the special usecase for Controller[0], which is sometimes hard-coded in the code.
Then calls like this are being made:
`getProtocolIndex(Settings.Protocol[0]);`
While something like this is to be expected:
`getProtocolIndex(Settings.Protocol[event->ControllerIndex]);`

psy0rz, 2018-01-10:
Ok i moved the mqtt stuff out of backgroundtasks. This fixes it in my scenario. (domotics MQTT with a message delay of 10000ms)

TD-er, 2018-01-10:
I've looked at the code-change.
Please also test with Domoticz MQTT as first (only?) controller and the switch and press it as fast as you can.
The yield is not always called, since sometimes the time between processing of messages is just about 1 sec and then it will not always call the yield.

psy0rz, 2018-01-10:
I did exactly that. Before it crashed immediately, and after the change it seems stable.

If the time is short enough you dont need to call yield. The watchdog will only kick in after 5 or 10 seconds or something.

psy0rz, 2018-01-10:
together with the memory issue fixes i consider this fixed for now. otherwise please comment/reopen. :)

commit 67fa39f4 (2018-01-10): moved mqtt stuff outside of backgroundtasks(). fixes #683 in my test scenario
closed 2018-01-10 by psy0rz