letscontrolit/ESPEasy #687: Error in device dropdown

jjansen85, 2018-01-09:
### Steps to reproduce
On dev-13; select _Communication - Kamstrup Multical 401 [TESTING]_ plugin

### Expected behavior
It should select _Communication - Kamstrup Multical 401 [TESTING]_

### Actual behavior
It selects _Display - HT16K33 [TESTING]_ plugin
In the source code it is clearly seen that both plugins use ID 57
```
<option value="57" selected="">Communication - Kamstrup Multical 401 [TESTING]</option>
<option value="57" selected="">Display - HT16K33 [TESTING]</option>
```

Software or git version:
Dev-13

psy0rz, 2018-01-10:
Where did you get "Communication - Kamstrup Multical 401 ".

You added that plugin yourself and then recompiled?

psy0rz, 2018-01-10:
I cant find that plugin anywhere.

closed 2018-01-10 by psy0rz

jjansen85, 2018-01-10:
No, it's part of dev-13, which I downloaded from the releases page.
In source it is _P071_Kamstrup401.ino

psy0rz, 2018-01-10:
If you downloaded it from the release page i should be able to find it ;) ill look agian.

psy0rz, 2018-01-10:
strange, now i found it. probably forgot to disable case sensitive when searching or something. 

Its indeed a bug (octal vs decimal notation.) will fix it now.

psy0rz, 2018-01-10:
done

psy0rz, 2018-01-10:
we just discovered that that plugin has a bug (we didnt notice it since the plugin wasnt actually used because of the previous bug.)

@TD-er is probably working on it as we speak.

jjansen85, 2018-01-11:
@TD-er just for my information / learning. I see you added this to the plugin
```
   case PLUGIN_WEBFORM_LOAD:
      {
        success = true;
        break;
      }
```
Why was the absence of this part causing problems? In previous releases, without this webform load it worked OK. So just wondering what has changed :-)

TD-er, 2018-01-11:
I am not entirely sure why it was working in the past. We (@psy0rz and I) tried to find the commit that could have changed this behavior, but failed miserably ;)
The problems it caused was simply that loading the plugin page never finishes and thus the watchdog timer kicks in and the device reboots.
As a result of this it was simply not possible anymore to enable or even delete the plugin.

So this fix looks simple, but had some major impact.
We also need to look into other plugins that may or may not have similar issues.

My suggestion is to simply not have defines for these case values, but use enums and then the compiler will warn when not all cases are being handled.

Apparently the plugin must return success = true, or else will retry.
And perhaps we should also limit the number or retries and also this bug was the reason to add issue #696 

So actually I am glad this bug appeared (now), since it will eventually result in better code :)

jjansen85, 2018-01-11:
Thanks for the clear explanation!

jjansen85, 2018-01-14:
Can confirm that the Kamstrup plugin is working like a charm again :)
(On Mega with ESPeasySoftwareSerial)