letscontrolit/ESPEasy #688: IRremoteESP8266 1.2.0 does not compile with core 2.4.0

s0170071, 2018-01-10:
... it cannot find the min() / max() functions. The bug is in the library and it is already fixed in the current verision as far as I can tell. 
The source of the error is that as of core 2.4.0 min() and max() functions force to have arguments of identical type which may require some casting. 
As this prevents compilation I suggest to update to the current version the library.
https://github.com/markszabo/IRremoteESP8266

s0170071, 2018-01-10:
@Grovkillen: its not really core related, more core triggered... sort of.
See devyte post here: https://github.com/esp8266/Arduino/issues/398
Solution is to update the lib.

Grovkillen, 2018-01-10:
I hear ya, I put it as only bug then :)

tarkan123, 2018-01-16:
I have integrated IRremoteESP8266 2.3.1 in to the latest Mega files (locally), making the necessary changes to the plugin files. Compiles successully.

However, since version 1.2.0 IRremoteESP8266, the number of supported protocols has grown, so not sure how this will impact memory and resources within ESPeasy.

Grovkillen, 2018-10-23:
Seems to be fixed?

closed 2018-10-23 by Grovkillen