letscontrolit/ESPEasy #691: Add support for Itead's SI7021 (acting as a 1-wire AM2301 sensor)

BRSS73, 2018-01-10:
The SI7021 (I2C sensor) from Itead is modified to act as a AM2301 (1-wire sensor, similar to DHT22) in order to be connected to Sonoff TH10/16 jack port (VCC, GND, NC, GPIO14).

ESPEasy (ESP_Easy_mega-20180104) can use DHT22, but cannot read the data ("DHT  : No reading!"), probably due to a protocol timing issue.

ESPurna firmware solve this issue about one month ago (https://bitbucket.org/xoseperez/espurna/issues/216/add-support-for-si7021) and it "seems" to be quite easy.

Can you implement this modification?

Thanks a lot.

Reference :
https://www.itead.cc/wiki/Sonoff_Sensor_Si7021
https://github.com/arendst/Sonoff-Tasmota/issues/735#issuecomment-348718383
https://bitbucket.org/xoseperez/espurna/commits/55ffb506ac611e0cc242a2d99dc303b2b155bcaa

BRSS73, 2018-01-13:
Works as expected with (at least) Itead's SI7021.

Thanks @jctual.

jctual, 2018-01-13:
Based on previous code, I made a separate plugin for SONOFF am2301 and si7021 to avoid to interfere with DHT 11/22.
I gave it number 90. Dont'know if it is right or not.
The code and the timing have been improved.
I tested it with two SONOFF am2301 and two SONOFF si7021. I got consistent readings and no protocol error, no checksum error.
Hope this may help.
[_P090_DHTSONOFF.zip](https://github.com/letscontrolit/ESPEasy/files/1628609/_P090_DHTSONOFF.zip)

BRSS73, 2018-01-14:
Your code can't compile as is. Boolean "waitState" is already defined in P005 as void.
Renamed "waitStateB" (or whatever else), it compile flawlessly and seems to run OK with my Si7021.

Regarding numbering, check this link: https://www.letscontrolit.com/wiki/index.php/ESPEasyDevelopmentGuidelines#Creating_new_plugins_on_the_playground

Thanks again.

jctual, 2018-01-14:
I have not this problem with waitState. It compiles fine for me.
You may have to remove the P005 I provided and replace by official P005.
After observing the results on a long period of time, I saw that am2301 sometimes gave strange results (+- 1° compared to normal value).
After reading carefully the datasheet of am2301, I saw that the init low pulse of 500µs required for si7021 is too low for am2301.
Two different values are required for am2301 (>800µs)  and si7021 (500µs). 
I modified the plugin to be able to select the sensor in the dropdown list.
It seems to work fine for both sensors
[_P090_DHTSONOFF.zip](https://github.com/letscontrolit/ESPEasy/files/1629308/_P090_DHTSONOFF.zip)
.
Thanks for your suggestion. I will look at the numbering issue and try to put this plugin in the playground.

BRSS73, 2018-01-14:
You're right about P005, my bad. I've added and compile your code in a wrong source directory.
In the good one, it's just fine.
Note to myself: do not test code at 2:20 am.

jctual, 2018-01-16:
There was a bug in my code. The variable "success" was not returned to true after reading the sensor.
This prevents the plugin to send data to domoticz.
Here the correct version.
Sorry for the inconvenience.
[_P090_DHTSONOFF.zip](https://github.com/letscontrolit/ESPEasy/files/1634562/_P090_DHTSONOFF.zip)

psy0rz, 2018-01-16:
I would prefer if the existing dht22 plug would support these sensors.

TD-er, 2018-01-16:
If there is a large overlap in code, perhaps the plugins should be merged in the same plugin, like I did with the BME/BMP280.

jctual, 2018-01-16:
I updated the original DHT plugin to include sonoff am2301 and si7021.
I tested it on am2301 and si7021.
The am2301 works as either DHT22 or SONOFF am2301.
I changed the code for reading the sensor to be more inline with am2301 datasheet and to avoid some spurious NAN readings.
It should work with DHT11 and DHT22 but I have no means to test it.
Any comments welcome
[_P005_DHT.zip](https://github.com/letscontrolit/ESPEasy/files/1635763/_P005_DHT.zip)

jctual, 2018-01-17:
oops .. DHT12 is missing.
This line 
if(Par3 == 11 || Par3 == 22)  delay(18);
should be replaced by this one
if(Par3 == 11 || Par3 == 22 || Par3 == 12)  delay(18);

I am not familiar with espeasy github.
I tried to create a new file on the github playground. But it disapeared just after.
Thanks for some hints to create a plugin on the github playground.

jctual, 2018-01-17:
Sorry to be back again.
Looking at different posts, it seems that the DHT plugin code was not correct.
The bus was pulled to high after the low init pulse. This may create a conflict since the sensor will pull the line down.
The actual code works but it's not safe.
It is better to remove the digitalwrite HIGH after the init low pulse.
I changed my code accordingly.

[_P005_DHT.zip](https://github.com/letscontrolit/ESPEasy/files/1639960/_P005_DHT.zip)

TD-er, 2018-01-17:
Is there a way how we can assist you in creating a fork + fixes and eventually a pull request?
It is great that others are willing to help, but working with zip files attached to replies is very hard to keep track of and to see what has been changed.

jctual, 2018-01-17:
I fully agree. It is not a good way.
I am completely new to git and github. I will dig into that later, but for the moment I have no time to do it properly.
It would be appreciated if you could tell me, or guide me to a tutorial, how to upload this plugin on github playground so that it can be merged later in the project.

Many thanks if you could help me.

shvarts1960, 2018-01-30:
Many thanks to jctual !
I replaced plugin with  yours, recompiled and now I have my sensor working.
But I see one strange thing. On "Wemos D1 mini" module everything works fine, but if I upload the same firmware into ESP-01 or ESP-12E module(s), I see in logs: 
DHT  : no Reading  !
DHT  : no Reading  !
...
DHT  : Temperature: 27.70
DHT  : Humidity: 18.50

I assume something wrong with timings?

TD-er, 2018-01-30:
Did you change something else, like cable length?
And it may very well be that some boards implement things differently with regard to pull-up resistors?

shvarts1960, 2018-01-30:
>Did you change something else, like cable length?
No, nothing.
Just now I simply reconnect sensor to other GPIO2 ( was at GPIO0 ), and things become much more better. From 10 readings only about 2 unsuccessfuly (no Reading).
It seems you are right about pull-up resistor value. I'll try to play with it.

TD-er, 2018-01-30:
Not all GPIOs are created equal. (some are more equal than others ;) )
If I remember correctly, GPIO-0 is also being used for sending out the wake pulse and toggle flash mode upon boot.
So better use other GPIOs.

jctual, 2018-01-30:
yes, probably a pullup issue.
To avoid conflicts on the bus and potential false readings, the signal pin is not actively driven to +Vcc.
A pullup resistor is mandatory as specified for these sensors.
This is not necessarily implemented on all boards and pins.
You should try 10k (down to 4.7k) between the signal pin and Vcc.
Take care also if the pin is not already used by another protocol (I2C, 1-wire) that may be activated in the firmware of your board. I had this issue on raspberry. Maybe this is also possible on ESP.

psy0rz, 2018-02-02:
@jctual i've created some documentation here for people that are new to github: https://www.letscontrolit.com/wiki/index.php/Tutorial_preparing_to_work_with_github

thanks for your trouble :)

uzi18, 2019-09-04:
you can try latest builds - AM2301 should be supported

thalesmaoa, 2021-11-27:
@uzi18 , with latest build i can't get am2301 to work. When I select SI7021 it works with some nan oscilating.

TD-er, 2021-11-27:
> @uzi18 , with latest build i can't get am2301 to work. When I select SI7021 it works with some nan oscilating.

What was the last build you tested where the sensor was working properly?

thalesmaoa, 2021-11-27:
Good question, so far:
Build: mega-20200204 - AM2301 (must be configured as SI7021) - Value and nan
Build: mega-20201227 - AM2301 (must be configured as SI7021) - Value and nan
Build: mega-20210114  - AM2301 (must be configured as SI7021) - Value and nan
Build:  mega-20210223 - AM2301 (must be configured as SI7021) - Value and nan
Build:  mega-20210503 - AM2301 (must be configured as SI7021) - When the value is not updated, the value is freezed (no update)
After that build: Value and nan

I have a fresh new TH16 / AM2301.
Looks like, none of them work with AM2301 config. Besides, NaN is common for all builds, except 20210503.

thalesmaoa, 2021-11-27:
After some tests I get:

```
639085 : Info   : DHT  : Temperature: 27.20
639086 : Info   : DHT  : Humidity: 70.20
639165 : Info   : EVENT: Sensor#Temperature=27.20
639265 : Info   : EVENT: Sensor#Humidity=70.20
644083 : Info   : DHT  : Protocol Timeout
649084 : Info   : DHT  : Protocol Timeout
```

tonhuisman, 2023-09-02:
This sensor is now supported in both I2C mode (P014) and 1-wire mode (P005), so this issue can be closed.

closed 2023-09-02 by TD-er