letscontrolit/ESPEasy #693: DS18B20 - possible timeout issues

rjblake, 2018-01-10:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Logging temperature every 60secs with DS18B20, DHT22 and BMP280

Does the problem presist after powering off and on? (just resetting isnt enough sometimes)
Yes

### Expected behavior
Tell us what should happen?
On occassion, I get a Temperature: Error! when trying to get temp from DS18B20 (mainly) and very occassionally from DHT22. No problem with BMP280

### Actual behavior
Tell us what happens instead?
e.g. Jan 10 19:08:10 ESP Unit: 1 : DS   : Temperature: Error! (28-ff-a3-76-c2-16-3-2c)

### System configuration
Using Rules to send Domoticz updates as MQTT causing other issues:
On System#Boot do    //When the ESP boots, do
  timerSet,1,60      //Set Timer 1 for the next event in 60 seconds
endon

On Rules#Timer=1 do
SendToHTTP,192.168.178.14,8084,/json.htm?type=command&param=udevice&idx=383&nvalue=0&svalue=[BMP280#Temperature];[BMP280#Pressure];BAR_FOR;ALTITUDE
SendToHTTP,192.168.178.14,8084,/json.htm?type=command&param=udevice&idx=384&nvalue=0&svalue=[DHT22#Temperature];[DHT22#Humidity];HUM_STAT
SendToHTTP,192.168.178.14,8084,/json.htm?type=command&param=udevice&idx=385&nvalue=0&svalue=[DSB18B20#Temperature]
  timerSet,1,60       //Resets the Timer 1 for another 60 seconds
endon

Hardware:
ESP8266 (Adafruit Huzzah Breakout) with DS18B20, DHT22, BMP280

Software or git version:
GIT version: | mega-01032018

TD-er, 2018-01-10:
What should the plugin do? Ignore error readings and not output any value if in error?
If I remember correctly, the DHT22 is quite a slow sensor and timeouts are allowed to happen.
But I can imagine sending values other than useful values may cause other issues, thus discarding these may sound like a good option to me.

rjblake, 2018-01-10:
The readings feed a script in Domotics with 'nil' values - guess I can trap them before they result in the script failing. The code has built in delays for the sensors, just thought it might be an option to allow these to be changed (e.g. perhaps 750ms in not enough for the DS at 12-bit resolution and could be solved with 800ms)

TD-er, 2018-01-10:
Well it is quite simple to change the plugin to only output valid readings.
Then you don't have an updated value every N seconds (set in the plugin), but sometimes just not an update.

psy0rz, 2018-01-10:
isnt this the same as #470

rjblake, 2018-01-11:
I'm not sure if it is the same, as the other issue relates to JSON output. I am returning values using rules and HTTP (as the Domoticz JSON is incorrect for the BMP280 and does not include all required info).

TD-er, 2018-01-11:
It is somewhat related in the sense of outputting invalid values.
When you stop doing that, it may also resolve the issues with the JSON formatter.
And I can imagine it could also cause issues further up the chain with tools processing those values.

psy0rz, 2018-01-11:
yeah we should just change invalid values to a valid null in json i think.

psy0rz, 2018-01-12:
fixed (should be in tonight's builds)

closed 2018-01-12 by psy0rz