letscontrolit/ESPEasy #694: Controller dropdown unsorted

Grovkillen, 2018-01-10:
We need to make the dropdown menu for controller settings sorted. It is now based on CXXX number.

psy0rz, 2018-01-10:
prefer to do that in 2.1 for now

psy0rz, 2018-01-10:
(it sorting seems a simple thing to do, but in c it could just as well introduce new issues)

Grovkillen, 2018-01-10:
I understand 👍

TD-er, 2018-01-11:
How do you want to have it sorted?
Based on protocol (MQTT/HTTP) or on receiver type (Domoticz/General/OpenHAB) or just alphabetical?

Grovkillen, 2018-01-11:
I was thinking alphabetical:

Domoticz HTTP
Domoticz MQTT
Generic HTTP
Generic MQTT
OpenHAB HTTP
OpenHAB MQTT
...
...

uzi18, 2018-01-11:
It is good idea to move sorting to JS, and take list with ajax api call

TD-er, 2018-01-11:
Hmm leaving processing to the browser is always nice.
Perhaps we could do similar things with real-time updates of the measurement data.

psy0rz, 2018-01-11:
yeah indeed, maybe we should consider moving even more things to the browser at some point. (e.g. the sorting of the devices combobox)

psy0rz, 2018-01-11:
in the best case scenario, espeasy would just serve cachable static html/css/js, and the api to configure espeasy would just need to be simple REST api. this solves lots of complexity and overhead in the code.

Grovkillen, 2018-08-04:
This will be handled by the browser in future GUI.

closed 2023-09-04 by TD-er