letscontrolit/ESPEasy #696: Add delete button and enable/disable option to device overview

TD-er, 2018-01-10:
The device overview screen only allows to enter the edit screen of the devices.
But what if one of the plugins is not working and crashing the ESP for whatever reason?
Then it would be nice to allow to disable or even delete the plugin without opening the plugin page.

Other ideas:

- Allow to enable/disable controllers.
- Allow to set controller id.
- Allow to change GPIO pins.

TD-er, 2018-01-13:
And a new idea: add an option to set the settings of a plugin to default. (see #712 )

Grovkillen, 2018-01-13:
Yes!!! Awesome.

Grovkillen, 2018-01-18:
Might be able to use the command "task Clear" for that button? 
https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference

TD-er, 2018-01-18:
I have to look into the code to see if that really does what the name suggests, but sounds like a good idea to use existing code :)

Grovkillen, 2018-01-18:
I use it "all the time" when deleting test devices :)

TD-er, 2018-01-18:
OK, then I will split the functionality into a separate function which may be called from the command interpreter and the delete button.
Or maybe just create a http link  (if that also works) to call the command functions.

Maybe also add a command to enable/disable a task.

Sounds like a good approach.

Grovkillen, 2018-01-18:
The TaskEnable,1 or TaskEnable,0 is indeed something I'd use.