letscontrolit/ESPEasy #698: Code cleanup

psy0rz, 2018-01-10:
There is still lots of dangerous code in ESPEasy.

I propose a generic code cleanup, this is a copy of a comment i made earlier in an older issue:

- use strlcpy instead of strncpy. strncpy is nasty, see https://linux.die.net/man/3/strncpy. When using strncpy, the size always needs to be one less than the buffer (sizeof()-1) , and you need to make the last byte always \0. We should probably replace all str*cpy stuff by strlcpy, or regular Strings.

- All the temporarty static log-buffers are dangerous: if we printf something to it that ends up too big after formatting we have a buffer overflow and crashes/instability.

- Use "IPAddress ip;" for ip-adresses and convert to string with ip.toString() when needed. 

- This gets copypasted all over the place and is horrible and unnecessary and dangerous:

```
strcpy_P(tmp, PSTR("HTTP : connecting to "));
sprintf_P(log, PSTR("%s%s using port %u"), tmp, host, ControllerSettings.Port);
```

Why not just:
```
sprintf_P(log, PSTR("HTTP : connecting to %s using port %u"),  host, ControllerSettings.Port);
```
Or better yet prevent the whole char log buffer:
```
addLog(LOG_LEVEL_DEBUG, String(F("HTTP : connecting to "))+host.toString()+":"+ControllerSettings.Port);
```

- addLog now also supports F() strings, so temporary buffers are no longer neccesary 

- This is something i also see a lot when reading network/http responses (also bad code that is copy/pasted a lot):

 ```
while (client.available()) {
    String line = client.readStringUntil('\n');
    line.toCharArray(log, 80);
    addLog(LOG_LEVEL_DEBUG_MORE, log);
    if (line.substring(0, 10) == F("HTTP/1.1 2")
```

Replace with :
```
  while (client.available()) {
    String line = client.readStringUntil('\n');
    addLog(LOG_LEVEL_DEBUG_MORE, line);
    if (line.startsWith(F("HTTP/1.1 2")))
    {
```

- I cleaned up DeleteNotNeededValues a bit with comments and extra variables to make it more readable. Keep in mind that hese variables dont take extra space. A String() created on one line also takes space. Usually the compiler can even optimze it better if you store stuff you repeat in extra variables. Main reason i cleaned it up is so we can better see if the function is safe or not.

TD-er, 2018-01-10:
Can even done with less code:
`addLog(LOG_LEVEL_DEBUG, String(F("HTTP : connecting to "))+ControllerSettings.getHostPortString());`

See #629

psy0rz, 2018-01-10:
ahhh right the nice new stuff you've added :)

TD-er, 2018-01-10:
Another thing that's bothering me is Strings are often already created and not being used.
For example for logs but the log level of all 3 log outputs is lower than the loglevel set in the addLog.

Also often you know how long a string is going to be, so do a reserve on it as soon as possible. That prevents heap fragmentation.

TD-er, 2018-01-11:
Another thing we should do is to use enums (instead of #defines) to reflect states like the plugin_load, _save, etc.
When using enums, the compiler can warn us when not all cases are being handled.

TD-er, 2018-01-11:
Also the ESPEasy.ino and Misc.ino are quite long. 
Should split based on tasks / structs / formatting functions etc.

Frequently used operations on structs should be part of the struct itself (object oriented), like I did with the ControllerSettings.

Also structs which are being stored, should be marked accordingly along with a description how much space is left to extend the struct and what the implications are when things get extended.
Also moving member variables around is a big no-no in those classes/structs, which should be mentioned accordingly. Perhaps store these in a separate file?

TD-er, 2018-01-11:
There is a lot of repetition of the same statements in the same file.
For example `[index - 1]` is being used 67 times in WebServer.ino.
Using a different name and don't repeat the "-1" would save a lot of chances for error.

s0170071, 2018-01-12:
There should also be some code addressing hardware faults like those failing flashes. I was thinking about CRC checks of code and spiffs. 
Checking the code should be done piece-wise in regular intervals.
I had a faulty configuration once and had a hard time getting rid of it because the Esp did not start properly.

psy0rz, 2018-01-12:
@s0170071 good point. at least we now check the error codes when reading/writing to flash and give feedback. (this wasnt even done as well before)

closed 2022-04-21 by TD-er