letscontrolit/ESPEasy #699: Title of device not correct

jjansen85, 2018-01-11:
### Steps to reproduce
Add a new device to the device page.

### Actual behavior
When I select for example plugin 3, pulse counter, it gives the following display
[image]
The settings are related to pulse counter, but the name is related to the next pluging (plugin 4)

On devices overview table name is correct

### System configuration
Software or git version:
GIT version: | v2.0-20180111

TD-er, 2018-01-11:
Looks like a 1-off-error :)

I'll look into it.

psy0rz, 2018-01-11:
Oops

TD-er, 2018-01-11:
Should be fixed in the next build.
Made a commit for the v2.0 branch and @psy0rz should merge it to Mega branch :)

closed 2018-01-11 by psy0rz