letscontrolit/ESPEasy #703: Make controller settings more "open" and API like

Grovkillen, 2018-01-12:
Current state is pretty closed down, the way ESP Easy talks to controllers are hard coded into each controller "plugin" file. As with the ever ongoing problem with Domoticz where some new sensors need to be reported using rules etc.

My idea is that we break these controller settings into PROTOCOL and CONTROLLER.

First you select..
Protocol: HTTP/MQTT
Then you select...
Controller: Generic/openHAB/Domoticz/etc.

And then you will be presented with a default setup page for that particular combination, in a kinda way that Generic HTTP Advanced work now.

This way it would be possible to have more control over the actual API between the systems without the need for custom compilation binary files. Easier to feed specific sensor values as a custom json or simple plain text etc. etc.

TD-er, 2018-01-12:
Another approach can be to create a "composite" plugin, which will take sensor values from other plugins. Just like the OLED and Framed OLED plugin and combine that to a single set of up to 3 values (the max number of values currently allowed per plugin) in any order desired.
Just like a normal plugin, this could then be coupled to up to 3 controllers.
Like this you can create any combination of sensor values a controller needs.
Especially Domoticz is quite peculiar on the set of values and its order to be used in certain templated views.

Then it is also possible to combine sensors (e.g. separate temp en humidity sensor) to a single controller message. (Domoticz needs all values of the same group in the same message)
And also adding a battery state value is then possible.

psy0rz, 2018-01-13:
@Grovkillen i dont think this is possible: I'm working on the testsuite, and the domotics http and mqtt protocols also seem to use different formats. So i dont think its possible to decouple controller from the underlying protocols, altough it would be nice if we could. :)

Grovkillen, 2018-01-13:
Thanks, we'll keep it as an open topic for the future?

psy0rz, 2018-01-13:
Ok