letscontrolit/ESPEasy #704: Add controller to upload to PVoutput.org

TD-er, 2018-01-12:
Issue #205 would be great to combine with upload to PVoutput.org.
To that site, you can upload the parameters read from these Eastron DIN rail meters and some more.

It would make a great addition to combine several of these meters + some weather information and upload to that controller.

There are more sites like this.
Maybe it is already possible to do using rules and Generic HTTP (Advanced???)

psy0rz, 2018-01-12:
should be possible with Generic http advanced i assume. Do you have link to the API?

TD-er, 2018-01-13:
@psy0rz https://pvoutput.org/help.html

wwpiotr, 2019-07-24:
Is it possible to configure ESPeasy to send data to pvoutput.org.
I have energy meter with pulse output, in espeasy I used Pulse counter.  Now i want to send data to pvoutput. 

Is any guide how to do it ?

uzi18, 2019-07-24:
take a look at http advanced wiki
simply add this controller to espeasy and click on question mark icon in his configuration

closed 2019-10-11 by Grovkillen