letscontrolit/ESPEasy #705: hardware menu broken

s0170071, 2018-01-12:
if I click on "hardware" :-1: 
[image]

maybe a chunking problem as I compiled with 2.4.0rc2

TD-er, 2018-01-12:
The HTML output for the Hardware page is about 7733 Bytes, so it is possible this is caused by the chunking.

s0170071, 2018-01-12:
... and it has those typical stray number(s).
Strange thing though is that I thought this was fixed- and in fact, the fix is still in place (webserver.ino). Is the hardware page generated elsewhere ?

TD-er, 2018-01-12:
Why is this labeled 2.0?
We don't support ESP core 2.4.x in ESPeasy v2.0

psy0rz, 2018-01-12:
right, dont compile espeasy v2.0 with core 2.4.0. Only the mega branch supports it for now. Should we make a note in the readme, or perhaps backport the support to 2.0?

closed 2018-01-12 by psy0rz

TD-er, 2018-01-13:
The only real important thing in the 2.4 branch is the WPA Krack fixes, as far as I know.

s0170071, 2018-01-14:
... and 2.4.0 feels like it is responding faster. I remember remotely I measured it once and it had like 40% faster web page delivery. But not sure about the figures any more. 
But the feature that I really do like (and use for my smart SML meter) is the configurable buffer of the hardware serial. 128 bytes was just too little.

TD-er, 2018-01-14:
In the Mega branch, I added a ESPeasy wrapper for the SoftwareSerial.
You can set the buffer size there.
It is no hardware serial, but at least it has interrupts tied to the pins, so it will not loose any data.