letscontrolit/ESPEasy #708: Update Process for 1MB Versions

florian-asche, 2018-01-12:
I did upgrade my devices to the newest version and the update failed. The device needs to be hard wired for an upgrade.

I did then read the notice on the wiki, that it no longer work with that small flash chips. 

How about checking the flash size before the flash process destroys the esp firmware. It should be solved with one small if function. Or disable the flash process from the beginning, instead show some info...

psy0rz, 2018-01-13:
Yeah, but that old firmware would have do the check. There is no way we can influence that.

Only thing we can do it add better documentation. Did you read anything anywhere, were we might add a warning? Perhaps the notes on the release page on github?

The warnings for upgrading from an old Rxxx version to ESPEasy v2.0:

- Only devices with 1Mb or more flash are supported.
- To use OTA you need more then 1Mb of flash. (otherwise your device will get unreachable via wireless and you need to fix it via serial)
- You lose all your config settings if you upgrade from an Rxxx version. (from before ESPEasy v2.0)

Grovkillen, 2018-01-13:
I'm planning on doing a netscanner app with OTA mass deploy tool where I'll add more info regarding this. Only windows though...

psy0rz, 2018-02-26:
@Grovkillen could you document my above comment on the proper places on the wiki?

Grovkillen, 2018-02-27:
Sure, will do

Grovkillen, 2018-02-27:
https://www.letscontrolit.com/wiki/index.php/ESPEasy#2.0.2B_firmware_on_1MB_modules

florian-asche, 2018-02-27:
no. wont do it.

florian-asche, 2018-02-27:
The upgrade programm should have a 
`if (memory => 1MB) { message "This upgrade progress cannot be used for upgrading device, Please use programming by cable." }`
This can be very usefull to help people avoid breaking their devices in future versions.

uzi18, 2018-02-27:
This project could considered to be useful for 1MB/8Mb modules to do OTA
updates with ESPEasy

https://github.com/soif/EspBuddy

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

florian-asche, 2018-02-27:
yes. but still, the standard upgrade process is not working for 1mb and should show a warning message, instead of break the device.
If that project is able to upload also 1mb OTA then we could add an override attribute.

Grovkillen, 2018-02-27:
Sure for now and future releases, but older releases don't have a check like that so it would still break those units.

TD-er, 2018-02-27:
Still adding a warning wouldn't hurt.
Will add it as suggestion to this issue: https://github.com/letscontrolit/ESPEasy/issues/882

Grovkillen, 2018-02-27:
Sure, but lets think about it. A generic warning that we could use for the day 4096 bytes aren't enough? We need to think ahead as well.

TD-er, 2018-02-27:
Just use some factor between free space and current sketch size. We'll think of something.
And I guess we're already past the 4096 bytes limit ;)

Grovkillen, 2018-02-27:
Haha yes.

uzi18, 2018-02-27:
Maybe it is possible to slit flash for OTA with  3/4 to 1/4 hah? 
Where last part 1/4 is minimal ota image to update real 3/4 flash.

TD-er, 2018-02-27:
Hmm that would be a bit tricky.
At least you should split it on multiples of flash pages (e.g. 512k and the rest)
And you have to indicate on upload what part of it to use.
Is it stored in SPIFF or on the final destination by the intermediate flash?

florian-asche, 2018-02-27:
i think adding a warning is a good start. But isnt your idea way to complicated?

I whould add some function that calculate the size that the ota file need. and then a second function that return the free space size. and then simply check if it fit or not.

TD-er, 2018-02-27:
@florian-asche Depends on how the intermediate flash firmware operates.
Does it write the file to SPIFF and then copy it to the final destination, or does it write it to the final sectors immediately?
I assume the first, since a lot can go wrong during transfer via WiFi.
So you have to know the free space of the intermediate file and the space of your own settings (don't want to overwrite those)

florian-asche, 2018-02-27:
I could live with lost settings. Add a pre warning step. Add a direct download link with settings backup for later upload. Maybe add some function to the upload process where the settings are written before the system reboots. Only some ideas, dont know if it works.

A complete other idea: Split the base system and all the sensor / actor files. The base system is very small and first time installed by serial. can be upgraded with ota because its size without all the sensors and actors is very small. All the sensors / actors come in packages and are opional installed by file upload or some download function. What packages are in use are saved in some config that will be redownloaded after 1mb ota update.

Grovkillen, 2018-03-05:
Lets reopen this?

Grovkillen, 2018-03-12:
Should we provide an official OTA bin file for this two stage update? I mean which is then maintained and released together with the rest of the bin files? Today we only have [this one](https://www.letscontrolit.com/wiki/index.php/ESPEasy#2.0.2B_firmware_on_1MB_modules) which will not be maintained.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er