letscontrolit/ESPEasy #709: Variable %systime% is not working in controller

PsychoAndi, 2018-01-12:
I use a Mini D1 ESP8266 module with two sensors, a BMP280 and a SI7021. The data is collected in a SQLite3 database on a Raspberry Pi3. Therefore i configured a controller with Generic HTTP. 

The Controller publish string is:
"wetterstation/sensorwrite.php?ssn=%sysname%&stn=%tskname%&svn=%valname%&sv=%value%&sntp=%systime%"

So far this is working except for "%systime%". 

User @TD-er has recognized the problem: In the code the "%systime%"-case is missing.

//TODO: move this to a generic replacement function?
url.replace(F("%sysname%"), URLEncode(Settings.Name));
url.replace(F("%tskname%"), URLEncode(ExtraTaskSettings.TaskDeviceName));
url.replace(F("%id%"), String(event->idx));
url.replace(F("%valname%"), URLEncode(ExtraTaskSettings.TaskDeviceValueNames[varIndex]));
if (longValue) 
  url.replace(F("%value%"), String(longValue));
 else
  url.replace(F("%value%"), toString(value, ExtraTaskSettings.TaskDeviceValueDecimals[varIndex]));

TD-er, 2018-01-12:
Fixing this will actually be part of #698 
There are several locations where system strings like these are being parsed.
So there should be a single function where all these strings should be handled at once.

PsychoAndi, 2018-01-13:
Thanks in advance!

For now i wrote a:
`url.replace(F("%lcltime%"), getDateTimeString('-', ':', ' '));`
in the right place and it works fine.

TD-er, 2018-01-21:
Should be fixed now.  (when merged)

closed 2018-01-22 by psy0rz